analogdevicesinc/msdk #907: Add target ID checking to OpenOCD Flashing

Brandon-Hurst, 2024-02-06:
The proposed enhancement is to add an ID check to the OpenOCD flashing routines (most likely in the scripts which call OpenOCD and give it commands). An OpenOCD script implementing ID check would...

- Reset Halt
- Read a target ID from a register on the MCU
- Verify that the target ID matches the _expected_ target ID
  - If a mismatch, alert the user to the fact that the .elf target does not match the connected target
  - Optionally direct them to the right location to re-configure their target .elf for the connected target
- Only attempt to flash the device if the above check is confirmed

This could potentially prevent attempting to flash mismatched targets (e.g. flash a .elf that is built for MAX32690 MCU to a MAX32666 MCU), which sometimes can cause situations where it becomes difficult to read/write the device's flash again.

EricB-ADI, 2024-02-07:
I have ran into this issue commonly and believe it is probably a good idea. @Jake-Carter have you messed with the flash scripts before?

Jake-Carter, 2024-02-07:
Agreed.  Did some digging and there is a "family code" entry in the USN from the flash info block we can use for this.

Just put this together for the MAX78000.cfg.  It will catch any mismatched connection and shut down the OpenOCD server.

```tcl
$_CHIPNAME.cpu configure -event reset-end {
	# The family code we expect to read from the MAX78000
	set FAMILY_CODE 0x85

	# Unlock the info block
	mww 0x40029040 0x3a7f5ca3 1
	mww 0x40029040 0xa1e34f20 1
	mww 0x40029040 0x9608b2c1 1

	# Read from info block.
	set info_block_pg1 [mrw 0x10800000]

	# Write to the info block again to re-lock it.
	mww 0x40029040 0xFFFFFFFF 1

	# Extract family_code.
	set usn_16 [expr { $info_block_pg1 >> 15 }]
	set usn_family_code [expr { ($usn_16 & 0xFF00) >> 8 }]
	if { [expr { $usn_family_code != $FAMILY_CODE }] } {
		echo "ERROR: Family code mismatch!  Expected $FAMILY_CODE but received 0x[format %X $usn_family_code]"
		echo "ERROR: Are you connected to the right microcontroller? (MAX78000)"
		shutdown 1
	}
}
```

Some more improvements are needed to improve the error messages and behavior.

On a mismatch, VS Code's flash tasks will look like this.  It will fail, but is not helpful.

```shell
 *  Executing task: arm-none-eabi-gdb --cd="/home/jhcarter/repos/msdk/Examples/MAX78000/Hello_World" --se="build/Hello_World.elf" --symbols=build/Hello_World.elf -x="/home/jhcarter/repos/msdk/Examples/MAX78000/Hello_World/.vscode/flash.gdb" --ex="flash_m4_run /home/jhcarter/repos/msdk/Tools/OpenOCD cmsis-dap.cfg max78000.cfg" --batch 

The target architecture is set to "armv7e-m".
Open On-Chip Debugger 0.11.0+dev-g56a818e4c (2023-10-09-16:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Remote communication error.  Target disconnected.: Connection reset by peer.
```

while `make flash.openocd` looks like this.  The "program" command somehow bypasses the shutdown or continuously reconnects.

```shell
~/repos/msdk/Examples/MAX78000/Hello_World (dev/micro-ros*) » make flash.openocd        jhcarter@jhcarter-X670E
Loaded project.mk
****************************************************************************
* Analog Devices MSDK
* - User Guide: https://analog-devices-msdk.github.io/msdk/USERGUIDE/
* - Get Support: https://www.analog.com/support/technical-support.html
* - Report Issues: https://github.com/Analog-Devices-MSDK/msdk/issues
* - Contributing: https://analog-devices-msdk.github.io/msdk/CONTRIBUTING/
****************************************************************************
Open On-Chip Debugger 0.11.0+dev-g56a818e4c (2023-10-09-16:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : CMSIS-DAP: SWD  supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 04231702dde703f600000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : max32xxx.cpu: Cortex-M4 r0p1 processor detected
Info : max32xxx.cpu: target has 6 breakpoints, 4 watchpoints
Info : max32xxx.cpu: external reset detected
Info : starting gdb server for max32xxx.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x100008b8 msp: 0x2001ffe0
Info : SWD DPIDR 0x2ba01477
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00002124 msp: 0x20003ff0
ERROR: Family code mismatch!  Expected 0x85 but received 0x84
ERROR: Are you connected to the right microcontroller? (MAX78000)
shutdown command invoked
Info : SWD DPIDR 0x2ba01477
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00002124 msp: 0x20003ff0
ERROR: Family code mismatch!  Expected 0x85 but received 0x84
ERROR: Are you connected to the right microcontroller? (MAX78000)
shutdown command invoked
** Programming Started **
Info : Padding image section 0 at 0x10008f24 with 12 bytes
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : SWD DPIDR 0x2ba01477
ERROR: Family code mismatch!  Expected 0x85 but received 0x84
ERROR: Are you connected to the right microcontroller? (MAX78000)
shutdown command invoked
shutdown command invoked
```