analogdevicesinc/msdk #913: Some questions about "fonts.h"

PYL4869, 2024-02-12:
Hi,

I plan to design a development board by myself using "MAX78000".  I noticed that "OV5640" was added to "Examples\MAX78000\CameraIF", so I updated all today using "MaintenanceTool.exe", but there were some problems.  For example, in "Examples\MAX78000\CameraIF" or "Examples\MAX78000\TFT_Demo", changing "board":"EvKit_V1" to "board":"FTHR_RevA" would cause this problem.
[image]

I noticed that "Libraries\MiscDrivers\Display" added "fonts". I'm not sure how this is causing the above problem, I tried to add a path but it didn't work. How do I fix this?
[image]

If I design "MAX78000" and use "OV5640", can I directly use "ov5640" related in "Libraries\MiscDrivers\Camera"?  What should I pay attention to?

spalani7, 2024-02-12:
Try including `Libraries/MiscDrivers/Display/fonts` directory in `<msdk>/Libraries/Boards/MAX78000/FTHR_RevA` - hopefully that should fix it.

berrystephenw, 2024-02-12:
I spent some time on a very similar error - right out of the box the feather wouldn't compile even the "hello world" once I changed the board type. 

@spalani7 suggested fix will probably work, but here is what I did:
I downloaded the latest MSDK from Git, and followed the instructions on cloning into a new directory and linking in the toolchain. I need to do that since I don't have a display and the latest code/makefiles allow you to turn that off. There are probably other fixes too that might be handy to have.

Jake-Carter, 2024-02-12:
As of #792 we replaced our TFT fonts with open source ones, and consolidated them to a single location.

The error above is happening because it looks like we pushed out an update to the MiscDrivers on Feb 1st (1.0.0-44) without also releasing the updated MAX78000 BSPs.

The key fix is to add [this](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Libraries/Boards/MAX78000/FTHR_RevA/board.mk#L73) line to the bottom of your MAX78000FTHR board.mk file.

I will request we push a release of the updated MAX78000 BSPs as well to hot-fix this issue

PYL4869, 2024-02-13:
> 从 #792 开始，我们用开源字体替换了 TFT 字体，并将它们合并到一个位置。
> 
> 发生上述错误是因为看起来我们在 2 月 1 日 （1.0.0-44） 推出了对 MiscDrivers 的更新，而没有发布更新的 MAX78000 BSP。
> 
> 关键的解决方法是将[此](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Libraries/Boards/MAX78000/FTHR_RevA/board.mk#L73)行添加到MAX78000FTHR board.mk 文件的底部。
> 
> 我将要求我们推送更新的 MAX78000 BSP 的版本，以热修复此问题

Hi
it is effective, but in "TFT Demo" this little error occurs.
[image]

In addition, if I design a  custom boards using "MAX78000", can I also use "board":"FTHR_RevA"?   Or should I refer to the "Custom BSP Template "section of the "MSDK User Guide"? Since I do not have "MAX78000EVKIT", I am not sure whether "OV5640" is used normally, For custom boards, can I directly use "ov5640" related in "Libraries\MiscDrivers\Camera"? What should I pay attention to?
[image]

Jake-Carter, 2024-02-14:
Hi @PYL4869, if you are going to use the TFT demos I would recommend following the instructions for [Developing from the Repo](https://github.com/Analog-Devices-MSDK/msdk?tab=readme-ov-file#developing-from-the-repo).  There are some others updates that have been made to the demos to use the new fonts.

Or you can start with this project.  We expect to have a release ready at the end of this week.
[MAX78000_TFT_Demo_Feb_2024_Pre-release.zip](https://github.com/Analog-Devices-MSDK/msdk/files/14274692/MAX78000_TFT_Demo_Feb_2024_Pre-release.zip)

Yes, for a custom board design, I would recommend following [Custom BSPs](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#custom-bsps).  The implementation will depend on the design of your board.  Have you completed the hardware design yet?  

In general, drivers are only added for hardware that is actually used.  Copying elements of the BSPs for our FTHR/EVKIT is not necessary unless you have copied elements of the hardware design.

PYL4869, 2024-02-14:
> Hi @PYL4869, if you are going to use the TFT demos I would recommend following the instructions for [Developing from the [...]

Hi
Thank you for your return on these issues.The hardware design is not yet complete. I'm also looking at the reference hardware designs "MAX78000FTHR" and "MAX78000EVKIT" and "MAXREFDES178". It seems trickier and more difficult than the "ST" or "Renesas" I'm familiar with. I may need more time to solve the hardware design of "MAX78000".

tyVandeZande, 2024-02-18:
Hey @Jake-Carter, The Custom BSP link gives me a 404 error. Will you please ensure it's public?
FYI I'm working with the MAX32690 EVKit

Thank you!

Jake-Carter, 2024-02-19:
Hi @tyVandeZande the new UG URL is https://analogdevicesinc.github.io/msdk//USERGUIDE/

Custom BSPs:  https://analogdevicesinc.github.io/msdk/USERGUIDE/#custom-bsps

See #923.  Unfortunately we've had to migrate our repo to the central ADI org and our old Github Pages domain is now broken...

tyVandeZande, 2024-02-19:
@Jake-Carter Thank you— I see that now!

Jake-Carter, 2024-02-19:
No problem, sorry for the trouble

PYL4869, 2024-04-04:
@Jake-Carter 
Hi, for a custom board design, I had some problems with my DAPLINK drive, you can refer to this [link](https://github.com/ARMmbed/DAPLink/issues/1070), about this [question](https://github.com/MaximIntegratedAI/MaximAI_Documentation/blob/main/MAX78000_Feather/README.md#how-to-unlock-a-max78000-that-can-no-longer-be-programmed), I may not be able to use "erase.act"allowing the device to be programmed again, Is there any way to unlock the non-programmable MAX78000?

closed 2024-04-09 by PYL4869