analogdevicesinc/msdk #915: osx-arm64 support

spalani7, 2024-02-12:
Currently I'm facing some issues trying to run a hello-world example for MAXIM 32672 EV Kit.

I initially faced issue #913 - solved it by fixing the board make file. Now that the build is passing - I tried to flash the output which resulted in OpenOCD complaining about not find the right libraries.

```
dyld[63421]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: <6EC14B58-C47C-3C9A-B797-B28F79542A85> ~/MaximSDK/Tools/OpenOCD/bin/openocd
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libhidapi.0.dylib' (no such file, not in dyld cache)
```

Seems like that problem is OpenOCD is looking for libraries for x86_64 rather that native architecture (arm64).

Jake-Carter, 2024-02-12:
Hi @spalani7, we're in the process of releasing ARM64 OpenOCD binaries for MacOS.  Previously, we've relied on Rosetta (see setup instructions [here](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#completing-the-installation-on-macos)) to run our OpenOCD binaries, but we received some reports last week that the procedure has changed on the latest MacOS.  Now we have the hardware and signing process setup for ARM64.

Here are the binaries that we'll be releasing in the next week or so.  If you wouldn't mind testing them for us and reporting any issues here we'd greatly appreciate it.

[openocd_0.12.0-1.0.0_latest_osx_arm64.zip](https://github.com/Analog-Devices-MSDK/msdk/files/14251334/openocd_0.12.0-1.0.0_latest_osx_arm64.zip)

spalani7, 2024-02-12:
Thanks @Jake-Carter - I totally missed those instructions on the userguide. Glad that I created this ticket anyways. Happy to test the above openocd and report if there are issues!

spalani7, 2024-02-14:
I can report that the above OpenOCD is working for me - feel free to close this once you have pushed a fix for this.

tyVandeZande, 2024-02-18:
@Jake-Carter Thank you for working on this. I tried installing the binaries above using the README instructions in the zip I'm receiving the same error as before installing the binaries:

`Creating /Users/tot/MaximSDK/Examples/MAX32690/Hello_World/build/max32690.hex
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 04091702d5aa035200000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : [max32xxx.cpu] Cortex-M4 r0p1 processor detected
Info : [max32xxx.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for max32xxx.cpu on 3333
Info : Listening on port 3333 for gdb connections
[max32xxx.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0x00000222 msp: 0xffffffd8
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000fff4 msp: 0x2000a900
** Programming Started **
Warn : Flash protection not supported on this device
Warn : Flash protection not supported on this device
Error: timeout waiting for algorithm, a target reset is recommended
Error: error -902 executing max32xxx flash write algorithm
Error: error writing to flash at address 0x10000000 at offset 0x00000000
** Programming Failed **
shutdown command invoked

Warn : Flash driver of max32xxx.flash does not support free_driver_priv()
Warn : Flash driver of max32xxx.flash1 does not support free_driver_priv()
make: *** [flash.openocd] Error 1`

Jake-Carter, 2024-02-19:
@EricB-ADI can you test a macOS OpenOCD flash for the MAX32690 against the error above?

EricB-ADI, 2024-02-19:
@Jake-Carter yes, I will. @EdwinFairchild  were you having a similar error on your laptop?

EdwinFairchild, 2024-02-19:
for me it was on ME17 I could never get it to succeed flashing

Jake-Carter, 2024-02-19:
@tyVandeZande Can you attach the current "DETAILS.TXT" (should be in the DAPLINK drive)?  Potentially related to #883 

Try updating your PICO firmware if you haven't already (see [here](https://analogdevicesinc.github.io/msdk/USERGUIDE/#max32625pico-pico))

tyVandeZande, 2024-02-19:
@Jake-Carter 
I have not updated the firmware. I will do that now. 

Pre-update TETAILS.txt: 

# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 04091702d5aa035200000000000000000000000097969906
HIC ID: 97969906
Auto Reset: 1
Automation allowed: 1
Overflow detection: 1
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0256
Git SHA: aab4d91774dc53be80fa4b48dc46e938b247ecec
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0x773b8887
Remount count: 0
URL: https://mbed.org/device/?code=@U?version=@V?target_id=@T

tyVandeZande, 2024-02-19:
@Jake-Carter @EricB-ADI 

I just updated the PICO. Same error. I'm not missing something simple, like putting the EVKit into a programming mode, correct? 
I have BOARD and TARGET set in project.mk —but nothing else? Am I missing criteria?

Updated Details.txt:
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-52-gb85b53df (armcc, local mods)
Unique ID: 04261702d5aa035200000000000000000000000097969906
HIC ID: 97969906
Auto Reset: 1
Automation allowed: 1
Overflow detection: 1
Incompatible image detection: 0
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0259
Git SHA: b85b53df17c24538eb1be57dcd08bb4553d09f9d
Local Mods: 1
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0xae88bc8d
Remount count: 0
URL: https://www.analog.com/en/design-center/evaluation-hardware-and-

Jake-Carter, 2024-02-19:
@tyVandeZande Setting TARGET and BOARD correctly (see [here](https://analogdevicesinc.github.io/msdk//USERGUIDE/#bsp-table)) should be it with the default jumper settings.

Can you run the following commands for me?

This will print the version number.

```shell
openocd --version
```

This one will log extra debug info about the problem to an `openocd_debug_log.log` file.  Ensure `$MAXIM_PATH` is set to the installed location of your MSDK before running.  Then please attach the log here as well.

```shell
openocd -d -s $MAXIM_PATH/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/max32690.cfg -c "init; reset halt" > openocd_debug_log.log 2>&1
```

tyVandeZande, 2024-02-19:
@Jake-Carter  openOCD version 0.12.0 

TARGET & BOARD are set correctly.

Here is the log:

`Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 4 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 4 options.c:52 configuration_output_handler(): 
Debug: 5 4 configuration.c:33 add_script_search_dir(): adding /Users/tot/MaximSDK/Tools/OpenOCD/scripts
Debug: 6 4 options.c:233 add_default_dirs(): bindir=/opt/homebrew/Cellar/open-ocd/0.12.0_1/bin
Debug: 7 4 options.c:234 add_default_dirs(): pkgdatadir=/opt/homebrew/Cellar/open-ocd/0.12.0_1/share/openocd
Debug: 8 4 options.c:235 add_default_dirs(): exepath=/opt/homebrew/Cellar/open-ocd/0.12.0_1/bin
Debug: 9 4 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 10 4 configuration.c:33 add_script_search_dir(): adding /Users/tot/Library/Preferences/org.openocd
Debug: 11 4 configuration.c:33 add_script_search_dir(): adding /Users/tot/.config/openocd
Debug: 12 4 configuration.c:33 add_script_search_dir(): adding /Users/tot/.openocd
Debug: 13 4 configuration.c:33 add_script_search_dir(): adding /opt/homebrew/Cellar/open-ocd/0.12.0_1/bin/../share/openocd/site
Debug: 14 4 configuration.c:33 add_script_search_dir(): adding /opt/homebrew/Cellar/open-ocd/0.12.0_1/bin/../share/openocd/scripts
Debug: 15 4 command.c:155 script_debug(): command - ocd_find interface/cmsis-dap.cfg
Debug: 16 7 configuration.c:88 find_file(): found /Users/tot/MaximSDK/Tools/OpenOCD/scripts/interface/cmsis-dap.cfg
Debug: 17 7 command.c:155 script_debug(): command - echo DEPRECATED! use 'adapter driver' not 'interface'
User : 18 7 command.c:685 handle_echo(): DEPRECATED! use 'adapter driver' not 'interface'
Debug: 19 7 command.c:155 script_debug(): command - adapter driver cmsis-dap
Debug: 20 7 command.c:155 script_debug(): command - ocd_find target/max32690.cfg
Debug: 21 7 configuration.c:88 find_file(): found /Users/tot/MaximSDK/Tools/OpenOCD/scripts/target/max32690.cfg
Debug: 22 7 command.c:155 script_debug(): command - reset_config srst_only
Debug: 23 7 command.c:155 script_debug(): command - adapter srst delay 2
Debug: 24 7 command.c:155 script_debug(): command - adapter srst pulse_width 2
Debug: 25 7 command.c:155 script_debug(): command - transport select swd
Debug: 26 7 command.c:155 script_debug(): command - ocd_find target/max32xxx.cfg
Debug: 27 8 configuration.c:88 find_file(): found /Users/tot/MaximSDK/Tools/OpenOCD/scripts/target/max32xxx.cfg
Debug: 28 8 command.c:155 script_debug(): command - ocd_find mem_helper.tcl
Debug: 29 8 configuration.c:88 find_file(): found /Users/tot/MaximSDK/Tools/OpenOCD/scripts/mem_helper.tcl
Debug: 30 8 command.c:155 script_debug(): command - add_usage_text mrw address
Debug: 31 8 command.c:155 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 32 8 command.c:155 script_debug(): command - add_usage_text mrb address
Debug: 33 8 command.c:155 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 34 8 command.c:155 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 35 8 command.c:155 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 36 8 command.c:155 script_debug(): command - ocd_find target/swj-dp.tcl
Debug: 37 8 configuration.c:88 find_file(): found /Users/tot/MaximSDK/Tools/OpenOCD/scripts/target/swj-dp.tcl
Debug: 38 8 command.c:155 script_debug(): command - transport select
Debug: 39 8 command.c:155 script_debug(): command - adapter speed 2000
Debug: 40 8 adapter.c:251 adapter_config_khz(): handle adapter khz
Debug: 41 8 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 42 8 adapter.c:215 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 43 8 command.c:155 script_debug(): command - transport select
Debug: 44 8 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 45 8 command.c:155 script_debug(): command - transport select
Debug: 46 8 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 47 8 command.c:155 script_debug(): command - transport select
Debug: 48 8 command.c:155 script_debug(): command - expr  [ string first "swd" $_TRANSPORT ] != -1 
Debug: 49 8 command.c:155 script_debug(): command - swd newdap max32xxx cpu -irlen 4 -ircapture 0x1 -irmask 0xf -ignore-version
Debug: 50 8 tcl.c:558 jim_newtap_cmd(): Creating New Tap, Chip: max32xxx, Tap: cpu, Dotted: max32xxx.cpu, 7 params
Debug: 51 8 core.c:1478 jtag_tap_init(): Created Tap: max32xxx.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 52 8 command.c:155 script_debug(): command - dap create max32xxx.dap -chain-position max32xxx.cpu
Debug: 53 8 command.c:155 script_debug(): command - target create max32xxx.cpu cortex_m -dap max32xxx.dap
Debug: 54 9 command.c:289 register_command(): command 'tpiu' is already registered
Debug: 55 9 command.c:289 register_command(): command 'rtt' is already registered
Debug: 56 9 command.c:155 script_debug(): command - max32xxx.cpu configure -work-area-phys 0x20005000 -work-area-size 0x8000
Debug: 57 9 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Debug: 58 9 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
[... cut ...]
Error: 82 17 cmsis_dap.c:293 cmsis_dap_open(): unable to find a matching CMSIS-DAP device
Debug: 83 17 command.c:545 run_command(): Command 'init' failed with error code -4
User : 84 17 command.c:608 command_run_line(): 
Debug: 85 17 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Warn : 86 17 core.c:224 flash_free_all_banks(): Flash driver of max32xxx.flash does not support free_driver_priv()
Warn : 87 17 core.c:224 flash_free_all_banks(): Flash driver of max32xxx.flash1 does not support free_driver_priv()
`

Jake-Carter, 2024-02-20:
@tyVandeZande looks like your `openocd` command may be going to your _system_-provided OpenOCD instead of ours.

`openocd --version` should print exactly this:

```serial
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
```

The .zip file I provided earlier puts the binary at `$MAXIM_PATH/Tools/OpenOCD/bin/openocd`.  Some of our environment setups expect it at `$MAXIM_PATH/Tools/OpenOCD/openocd`.  I've fixed that in our installer packages, but in the meantime you can run the following command to symlink it at the expected location.

```shell
ln -s $MAXIM_PATH/Tools/OpenOCD/bin/openocd $MAXIM_PATH/Tools/OpenOCD/openocd
```

Then the [command-line setup instructions](https://analogdevicesinc.github.io/msdk/USERGUIDE/#setup-command-line) should work for configuring the PATH correctly to load our openocd first.  Afterwards, your `openocd --version` string should match exactly, and I would expect the flash driver issues to go away.

tyVandeZande, 2024-02-22:
@Jake-Carter 

I tried this initially, and when I added this link, it returned the result `File Exists` but whenever I ran `flash.openocd` or `openocd --version` it would return `no such file or directory` I also tried using a zprofile alias to that location. So I tried installing the conventional openOCD package. 

Is there a step-by-step video somewhere to follow? 
Or a way to install all requirements, without using the automatic installer?

I tried removing and re-installing the MaximSDK (and all links/ profile entries) from the very beginning, using the automatic installer. Now, when I try to run `setenv.sh` an error returns in `updates.sh`  and says `MaintenanceTool` doesn't exist— however it is located exactly where `updates.sh` is looking.

Thanks!

Jake-Carter, 2024-02-22:
> I tried this initially, and when I added this link, it returned the result File Exists but whenever I ran flash.openocd or openocd --version it would return no such file or directory I also tried using a zprofile alias to that location. So I tried installing the conventional openOCD package.

> Is there a step-by-step video somewhere to follow?
> Or a way to install all requirements, without using the automatic installer?

We run a custom fork of OpenOCD, so it's required to use ours to interface with our parts.  The system-provided OpenOCD is missing flash drivers, among other things.  The automatic installer is the only way we distribute our binaries.  FWIW we have pushed our release to the installer, so it should now be pulling the latest resources as of this morning and doing the heavy lifting for you.

It sounds like you are just having PATH issues.  The `updates.sh` error is a strange one...  Try deleting the file, or moving it somewhere else for now.  `setenv.sh` will only attempt to run it if it exists.

That should at least get your `setenv.sh` script to complete.  Afterwards you can try `openocd --version` again.  If you're still getting failures, try `echo $PATH`.  It should look something like this after running `source setenv.sh`...

```shell
/home/jhcarter/repos/msdk/Tools/GNUTools/Make:/home/jhcarter/repos/msdk/Tools/OpenOCD:/home/jhcarter/repos/msdk/Tools/xPack/riscv-none-elf-gcc/12.2.0-3.1/bin:/home/jhcarter/repos/msdk/Tools/xPack/riscv-none-embed-gcc/10.2.0-1.2/bin:/home/jhcarter/repos/msdk/Tools/GNUTools/10.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
```

... notice the toolchain locations pre-pended to the beginning of your PATH variable.  This forces our tools to load before any system-provided ones.

By the way - what environment are you using (i.e. VS Code, Eclipse, command-line)?
> I tried removing and re-installing the MaximSDK (and all links/ profile entries) from the very beginning, using the automatic installer. Now, when I try to run setenv.sh an error returns in updates.sh and says MaintenanceTool doesn't exist— however it is located exactly where updates.sh is looking.

`MaintenanceTool` not existing is a weird one...  it could be a permissions issue. 

What does `ls -la` show from inside the root MaximSDK install folder, and the `Tools/OpenOCD` folder?

The root folder should look something like this:
[image]

and `Tools/OpenOCD` should look like this, with the `bin` folder and `openocd` symlink pointing inside of it.
[image]

tyVandeZande, 2024-02-23:
@Jake-Carter 

I will respond more fully soon, but I cannot see any substantial differences between your screenshots and my file structure. There may be permission issues adding to the complications— When I used the .dmg automatic installer, I was required to "Open Anyway" from system preferences, twice. 

When I ran `openocd --version` this time, I received `zsh: permission denied: openocd`
when I sudo'd the above I received `sudo: openocd: command not found`

I'm working through the command line.

I will try a full reinstall on a fresh user profile and follow up. 
I'm currently running Ventura 13.2.1 on a 2021 M1

Thanks for all your help.

tyVandeZande, 2024-02-23:
Hi @Jake-Carter I ran into problems with the fresh install when I checked the validity of the tools.
Here is the full terminal window:

```
Last login: Thu Feb 22 22:41:16 on ttys000
/Users/code/MaximSDK/updates.sh: line 8: /Users/code/MaximSDK/MaintenanceTool: No such file or directory
code@Tys-MacBook-Pro ~ % arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/Users/code/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: /tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native --libexecdir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/lib --infodir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-mpfr=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-mpc=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-isl=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-libelf=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Arm Embedded Toolchain 10.3-2021.10' --with-multilib-list=rmprofile,aprofile
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10) 
code@Tys-MacBook-Pro ~ % arm-none-eabi-gdb -v
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
code@Tys-MacBook-Pro ~ % make -v
GNU Make 4.4.1
Built for x86_64-apple-darwin19.6.0
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
code@Tys-MacBook-Pro ~ % openocd -v
dyld[13808]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: <6EC14B58-C47C-3C9A-B797-B28F79542A85> /Users/code/MaximSDK/Tools/OpenOCD/i386/bin/openocd
  Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file, not in dyld cache)
zsh: abort      openocd -v
code@Tys-MacBook-Pro ~ % 
```

tyVandeZande, 2024-02-23:
@Jake-Carter I just tried adding the binaries from above and adding the symlink and received the same `zsh: permission denied: openocd` and 'sudo: openocd: command not found' — I just double checked `ls -la` on the fresh user and it is the same as yours. The file is not being recognized. 

@spalani7 Will you please let us know two pieces of info: 
A. Which macOS version are you using? B. Do you have FileVault turned on?

Thank you!

tyVandeZande, 2024-02-23:
@Jake-Carter I upgraded my machine to Sonoma 14.3.1 with no success. Which macOS are you using?

`ln: /Users/tot/MaximSDK/Tools/OpenOCD/openocd: File exists` still returns when I symlink, but when I call `openocd` it says `command not found` I will post to Apple developer forms (not mentioning this specific issue) to see if others are having issues with running binaries through terminal.

Jake-Carter, 2024-02-23:
> Hi @Jake-Carter I ran into problems with the fresh install when I checked the validity of the tools. I tried this while running Terminal using Rosetta2. I did not switch the default openOCD files to the zip above.
> 
> Here is the full terminal window:
> 
> ```
> Last login: Thu Feb 22 22:41:16 on ttys000
> /Users/code/MaximSDK/updates.sh: line 8: /Users/code/MaximSDK/MaintenanceTool: No such file or directory
> 
> code@Tys-MacBook-Pro ~ % arm-none-eabi-gcc -v
> Using built-in specs.
> COLLECT_GCC=arm-none-eabi-gcc
> COLLECT_LTO_WRAPPER=/Users/code/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/lto-wrapper
> Target: arm-none-eabi
> Configured with: /tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native --libexecdir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/lib --infodir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-mpfr=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-mpc=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-isl=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-libelf=/tmp/jenkins-GCC-10-pipeline-338_20211018_1634516195/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Arm Embedded Toolchain 10.3-2021.10' --with-multilib-list=rmprofile,aprofile
> Thread model: single
> Supported LTO compression algorithms: zlib
> gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10) 
> 
> code@Tys-MacBook-Pro ~ % arm-none-eabi-gdb -v
> GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
> Copyright (C) 2021 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.
> 
> code@Tys-MacBook-Pro ~ % make -v
> GNU Make 4.4.1
> Built for x86_64-apple-darwin19.6.0
> Copyright (C) 1988-2023 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.
> 
> code@Tys-MacBook-Pro ~ % openocd -v
> dyld[13808]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
>   Referenced from: <6EC14B58-C47C-3C9A-B797-B28F79542A85> /Users/code/MaximSDK/Tools/OpenOCD/i386/bin/openocd
>   Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file, not in dyld cache)
> zsh: abort      openocd -v
> 
> code@Tys-MacBook-Pro ~ % 
> ```
> 
> Same error that started this thread. I tried to install `hidapi` but it was already installed and up to date. Could all this be a result of having macOS FileVault turned on?

At this point the only error you have is a missing search path for the libraries.  (Duplicate of https://github.com/analogdevicesinc/VSCode-Maxim/issues/21)  It seems that Homebrew will install `hidapi` to `/opt/homebrew/Cellar/hidapi` on some systems and fail to symlink it properly.

A reported workaround is to manually symlink `/usr/local/opt/hidapi/` -> `/opt/homebrew/Cellar/hidapi`.  You could also try `brew link hidapi`.  I believe that command is intended to have the same effect, and it should have been run on the installation of the library.  This may be a bug in Homebrew's formulae for `hidapi` that we need to report.

Your latest errors could also be similar dependency issues (see https://stackoverflow.com/questions/1562071/how-can-i-find-which-elf-dependency-is-not-fulfilled)

Sorry for all the trouble...  This has the symptoms of the usual dependency hell that can sometimes happen on Linux/MacOS.  What I usually do to troubleshoot this is use `ldd` to list the dependencies a binary is looking for, and where.

i.e. on Linux:

```shell
~/MaximSDK/Tools/OpenOCD/bin » ldd openocd
	linux-vdso.so.1 (0x00007ffc1a9cb000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fb793586000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb793581000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb79357c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb792c00000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fb7935da000)
```

I'm mostly a linux user... We are limited in our MacOS hardware at the moment, so we're reliant on the community.  When we find a resolution to your issue we will integrate it into our installer/documentation for MacOS users.

spalani7, 2024-02-23:
> @Jake-Carter I just tried adding the binaries from above and adding the symlink and received the same `zsh: permission denied: openocd` and `sudo: openocd: command not found` — I just double checked `ls -la` on the fresh user and it is the same as yours. The binary is not being recognized.
> 
> @spalani7 Will you please let us know two pieces of info: A. Which macOS version are you using? B. Do you have FileVault turned on?
> 
> Thank you!

my macos version is 13.4.1 and yes my file vault is ON.

Eric-Busch0, 2024-02-23:
@Jake-Carter I do not see these issues on my system flashing MAX32690. That is all I have at home.

tyVandeZande, 2024-02-24:
@Jake-Carter !success! ....it just require the time to be 1:30 am for it to work. I will follow up soon. Of course it was a small misunderstanding on my part. thx

Jake-Carter, 2024-02-26:
Thanks @tyVandeZande, let me know what the issue was when you can

Jake-Carter, 2024-02-27:
Closing out - the updated MacOS binaries have been released as of https://github.com/analogdevicesinc/msdk/releases/tag/v2024_02

closed 2024-02-27 by Jake-Carter

tyVandeZande, 2024-02-29:
> Thanks @tyVandeZande, let me know what the issue was when you can

The issue was the temp README instructions in the zip file above. I likely over-wrote a file that was needed. In that process I ran into this error https://github.com/analogdevicesinc/VSCode-Maxim/issues/21#issue-2076308269 — So I think this stems from overwriting necessary files.

Thanks