analogdevicesinc/msdk #920: make clean.cordio does not work

EricB-ADI, 2024-02-14:
It appears the full path to libraries is not being appended to when using make clean.cordio

```
make[1]: Entering directory '/home/eric-busch/Workspace/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
/home/eric-busch/Workspace/msdk/Libraries/Cordio/platform/targets/maxim/build/libCordio.mk:94: /Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655.mk: No such file or directory
make[1]: *** No rule to make target '/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655.mk'.  Stop.
make[1]: Leaving directory '/home/eric-busch/Workspace/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
make: *** [/home/eric-busch/Workspace/msdk/Libraries/Cordio/platform/targets/maxim/build/cordio_lib.mk:100: clean.cordio] Error 2

```

Jake-Carter, 2024-02-14:
Seems to work OK for me on the latest main.  Tested on Windows as well.  This has the symptoms of some older MAXIM_PATH issues that should be fixed...  do you have the latest main checked out?

```shell
~/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats (main*) » make clean.cordio
Loaded project.mk
---INFO enter Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk, RISCV_CORE=, RISCV_LOAD=
---INFO ARM ONLY, SRAM_SIZE=0x20000
---INFO exit  Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk
****************************************************************************
* Analog Devices MSDK
* - User Guide: https://analog-devices-msdk.github.io/msdk/USERGUIDE/
* - Get Support: https://www.analog.com/support/technical-support.html
* - Report Issues: https://github.com/Analog-Devices-MSDK/msdk/issues
* - Contributing: https://analog-devices-msdk.github.io/msdk/CONTRIBUTING/
****************************************************************************
make[1]: Entering directory '/home/jhcarter/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
---INFO enter Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk, RISCV_CORE=, RISCV_LOAD=
---INFO ARM ONLY, SRAM_SIZE=0x20000
---INFO exit  Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk
- RMDIR /home/jhcarter/repos/msdk/Libraries/Cordio/bin/max32655_baremetal_T1_softfp
make[1]: Leaving directory '/home/jhcarter/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
```

EricB-ADI, 2024-02-14:
Hmm. I recently unset my MAXIM_PATH because of issues I had with switching between different repos. Does this not use the relative path?

Jake-Carter, 2024-02-14:
Should also work with the relative path.  Do you get that warning line showing you what relative path it calculates?

```shell
~/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats (main*) » make clean.cordio                         jhcarter@jhcarter-X670E
Loaded project.mk
Makefile:106: Warning:  MAXIM_PATH is not set!  Set MAXIM_PATH in your environment or in project.mk to clear this warning.
Makefile:107: Warning:  Attempting to use /home/jhcarter/repos/msdk calculated from relative path
---INFO enter Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk, RISCV_CORE=, RISCV_LOAD=
---INFO ARM ONLY, SRAM_SIZE=0x20000
---INFO exit  Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk
****************************************************************************
* Analog Devices MSDK
* - User Guide: https://analog-devices-msdk.github.io/msdk/USERGUIDE/
* - Get Support: https://www.analog.com/support/technical-support.html
* - Report Issues: https://github.com/Analog-Devices-MSDK/msdk/issues
* - Contributing: https://analog-devices-msdk.github.io/msdk/CONTRIBUTING/
****************************************************************************
make[1]: Entering directory '/home/jhcarter/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
---INFO enter Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk, RISCV_CORE=, RISCV_LOAD=
---INFO ARM ONLY, SRAM_SIZE=0x20000
---INFO exit  Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/max32655_memory.mk
- RMDIR /home/jhcarter/repos/msdk/Libraries/Cordio/bin/max32655_baremetal_T1_softfp
make[1]: Leaving directory '/home/jhcarter/repos/msdk/Examples/MAX32655/Bluetooth/BLE_dats'
```

closed 2024-03-22 by Jake-Carter