analogdevicesinc/msdk #921: Support for CMake

spalani7, 2024-02-14:
Is there any plan to include CMake modules to help find right board and device files for upstream application project that is based on CMake ?

I'm planning to have an application code which is based on CMake build system to help generate necessary project files like  makefiles, IAR, or Keil IDE's. Just trying to take IDE agnostic approach. In order to point my CMake build system to right BSP and DFP packages it gets tricky to use CMake to read makefiles from SDK.

Since I dont see specific BSP and DFP's hosted in analog.com - I just wanted to check if there is a CMake module that implements [find_package](https://cmake.org/cmake/help/v3.28/command/find_package.html) or similar that will help upstream node to fetch the list of files/directories to include in the build for any given specifc board or target.

Jake-Carter, 2024-02-15:
Not at the moment, no.

My opinion on CMake is that it's great for large cross-platform projects where you're building for a traditional host PC.  For our embedded development most of its features are not useful since we are providing the entire toolchain anyways.  We need more control over the IDE generators to get them to work, and the custom scripting language is a bit clunky to work with at the project-specific level without some of the features we've implemented in our build system like autosearch.  There are some deeper limitations as well when we get to the link step and some of the custom .elf stitching we're doing for dual-core micros.

It might make more sense for our libraries eventually, but right now we provide direct support for VS Code, Eclipse, Keil, and IAR and our Makefiles are cross-platform.

spalani7, 2024-02-15:
Thanks Jake for prompt answer.
One last question - is there any other source for BSP and DFP's other than SDK at this point ?
If SDK is the only source - is it reasonable to assume the support for both BSP's and DFP will be through SDK for long time and the directory search would be mostly frozen ?

Jake-Carter, 2024-02-15:
For BSPs - no, the SDK is the only source for our evaluation platforms.  IAR/Keil have their own delivery mechanisms for the BSPs, but they are pulled from the exact same source and directory structure.

Could you clarify what a DFP is?

In general our directory structure is frozen.  The only thing that changes somehat frequently is the Examples folder, but that shouldn't affect builds.

spalani7, 2024-02-18:
DFP - stands for - Device Family Pack.

Example: For AD8CM302x family 

- DFP: http://download.analog.com/tools/EZBoards/CM302x/Releases/AnalogDevices.ADuCM302x_DFP.3.2.0.pack
- BSP: http://download.analog.com/tools/EZBoards/ADICUP3029/Releases/AnalogDevices.EVAL-ADICUP3029_BSP.1.1.0.pack

closed 2024-02-18 by spalani7