analogdevicesinc/msdk #923: 404 to github-pages

remren, 2024-02-15:
Sorry to bother y'all with this, but for some reason the [https://analog-devices-msdk.github.io/msdk/](https://analog-devices-msdk.github.io/msdk/) site is down with a 404 error. I recall it was working a few days ago, so not sure if the recent deployment may have broken something.

The other site [https://analogdevicesinc.github.io/MAX32xxx-MSDK/](https://analogdevicesinc.github.io/MAX32xxx-MSDK/) seems to work just fine.

Jake-Carter, 2024-02-16:
Thanks @remren, we're in the process of a required migration of our repos to the central [Analog Devices org](https://github.com/analogdevicesinc).

Unfortunately the migration will break our Github Pages URLs...  We can't keep an older placeholder around because then the repo auto-forwarding after the transfer won't work...  We can't get the old URL because the domain is owned by Github...  So sadly it's an unavoidable break.  We'll be updating all existing URLs in our code/documentation once the migration is complete

Jake-Carter, 2024-02-16:
Final move is complete.  The new URL is now https://analogdevicesinc.github.io/msdk/

closed 2024-04-16 by Jake-Carter