analogdevicesinc/msdk #936: Not restarting advertisement after stopped. FreeRTOS tickless

EdwinFairchild, 2024-02-27:
Resuming advertising is not  working once they are stopped. If you comment out line `230` and `239`, essentially remove the if statement and thus the call to `PalBbGetTimestamp` ,  and  keep the if statements contents then it works just fine. 
I think it may be related to accessing the dbb but the `PalBbGetTimestamp` guards against that in the phy by checking `palBbEnableCnt`

https://github.com/analogdevicesinc/msdk/blob/47ac9e535d29921286969f6e486b673ee6ff9d56/Examples/MAX32665/Bluetooth/BLE_FreeRTOS/freertos_tickless.c#L228-L240

### Steps to reproduce:
- Build BLE_FreeRTOS example with tickless mode enabled
- Start application
- Use phone app to see advertising works fine
- [ Button 2 press short ] will stop advertising
- [ Button 1 press short ] will resume advertising (does not resume)

EricB-ADI, 2024-04-10:
Not having an issue with it.

EricB-ADI, 2024-04-10:
Removing the if statement basically causes it to never go to sleep.

EricB-ADI, 2024-04-10:
BB was not being correctly disabled and enabled/restored.

closed 2024-05-17 by EricB-ADI