analogdevicesinc/msdk #947: Incorrect pre-scaler used for ME10 continuous timer when a prescaler of 4096 is chosen

emmet-horgan, 2024-03-07:
I noticed that when setting up timers using the `MXC_TMR` API, targeting the MAX32650, that when a prescaler of 4096 is chosen, it does not actually take effect for the timer. For example, the following code snippet,
```
    mxc_tmr_cfg_t tmr;
    MXC_TMR_Shutdown(MXC_TMR0);
    tmr.pres = TMR_PRES_4096;
    tmr.mode = TMR_MODE_CONTINUOUS;
    tmr.cmp_cnt = 0xFFFFFFFF; // Maximum 32 bit value
    tmr.pol = 0;
    
    MXC_TMR_Init(MXC_TMR0, &tmr);
    MXC_TMR_Start(MXC_TMR0);
```
initializes the timer with a pre-scaler of 4096, but in reality the timer counter is incremented at the `PeripheralClock` rate (60 MHz for the MAX32650) meaning that a pre-scaler of 1 was applied. I was able to rudimentally verify the pre-scaler was actually 1 by measuring the time it takes to overflow the `tmr.cmp_cnt` register (~70 seconds). I checked the initialization through a debugger and everything appeared to be perfect with all the right values set for a pre-scaler of 4096. The timer appears to have the correct frequency for other pre-scalers, I've only noticed the issue with 4096 in particular. The issue persists for `MXC_TMR0` and `MXC_TMR1` but I haven't verified for the other timers. I'm not sure if the issue is software, or hardware, I have verified that the issue occurs for both the [AD-SWIOT1L-SL](https://www.analog.com/en/resources/evaluation-hardware-and-software/evaluation-boards-kits/ad-swiot1l-sl.html#eb-documentation) and the FTHR_APPS_A boards.

Jake-Carter, 2024-03-08:
Thanks @emmet-horgan, I will check with design to see if there are any known hardware issues at a prescaler of 4096.  

The software seems correct based on the MAX32650 UG, but this TMR hardware is an older design that splits out the MSb of the divider into its own bit.  There may be a hardware issue at this setting.

Jake-Carter, 2024-03-27:
Hi @emmet-horgan sorry for the radio silence...  took a bit to check through since this is an older part, but this doesn't seem to be a known issue that's logged anywhere.  Surprising that no one has encountered it yet.

We'll do some more testing across different clock sources and boards, and if we can replicate it across all of them I think it's best we remove 4096 as a prescaler option

emmet-horgan, 2024-03-28:
Okay interesting. Were you able to reproduce the error by any chance ?

Jake-Carter, 2024-03-28:
Yes, @selimgullulu was able to replicate the issue on the 32651, so we can confirm you've found an issue universal to the ME10.  It also seems there might be some issues with the 512, 1024, and 2048 pre-scalers.  Did you see this as well?

[MAX32651_TMR_Example_Test_Results.xlsx](https://github.com/analogdevicesinc/msdk/files/14793663/MAX32651_TMR_Example_Test_Results.xlsx)

selimgullulu, 2024-03-29:
Hi all, the problem seems to be due to a division shift in the tmr_reva.c file.
Jake, I added a comment to that file in your branch https://github.com/Jake-Carter/msdk/tree/fix/gh-947
I tested the tmr example with the updated code and it worked for all the following prescalers (128,256,512,1024,2048 and 4096)

Jake-Carter, 2024-04-01:
Thanks @selimgullulu 

Looks like this was a mistake in our UG that made it through to the drivers.  Updated the PR, @emmet-horgan it should fix the issue

emmet-horgan, 2024-04-02:
Thanks for the support

PR #971 (2024-04-03): fix(PeriphDrivers): Fix Bitwise Logic for TMR RevA Prescaler Init
### Description

Fixes #947 

The ME10 UG has a mistake that propagated itself to our periphdrivers.  The logic table skips from `0b000` to `0b010` at the transition between 256 -> 512.  This causes 512 - 2048 to be twice what they should, and 4096 to roll over to an invalid setting that behaves like a divisor of 1.

[image]

This PR updates the divisor settings to remove the skip

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [... cut]
closed 2024-04-03 by Jake-Carter