analogdevicesinc/msdk #948: Minor warnings at mxc_sys.h

alperenguclu, 2024-03-08:
I am receiving following warnings in the mxc_sys.h file, and this issue seems to be present in most microcontrollers.

- mxc_sys.h:220:20: warning: function declaration isn't a prototype [-Wstrict-prototypes]

  220 | static inline void _mxc_crit_get_state()

- mxc_sys.h:351:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]

  351 | int MXC_SYS_RTCClockDisable();

Jake-Carter, 2024-03-08:
Which GCC version are you compiling with?

alperenguclu, 2024-03-11:
arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

Jake-Carter, 2024-03-11:
Strange, `-Wstrict-prototypes` shouldn't be enabled by default...  I don't see the warning on my machine or any of the build auto-testers, but if I manually enable it I see tons of warnings on other functions as well.

Does the flag show up explicitly in a verbose build for you?  You can check with grep:

```shell
make distclean && make VERBOSE=1 | grep -- "-Wstrict-prototypes"
```

The fix is pretty trivial, we just need to add `void` to any functions with empty args.  i.e. 

```diff
-static inline void _mxc_crit_get_state()
+static inline void _mxc_crit_get_state(void)
```

Just curious how/why you're seeing the flag enabled.

alperenguclu, 2024-03-13:
Yes, Jake, we are working on a Security project that will be integrated to Maxim SDK and we are using the -Wstrict-prototypes. I guess it will be nice to add void to functions with no args.

PR #954 (2024-03-20): fix(Build): Fix `-Wstrict-prototypes` warnings
## Pull Request Template

### Description

Closes #948.  Essentially this adds and enforces `int myfunction(void)` instead of `int myfunction()`

Also merges the release branch again so that #937 should start working properly.  The 2024 release tag doesn't seem visible from the main branch yet, but merging release locally fixes it.

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related GitHub issues [using a keyword](http [... cut]
closed 2024-03-20 by sihyung-maxim