analogdevicesinc/msdk #949: MAX32690/ME18 Alternate I2C

BrentK-ADI, 2024-03-08:
The ME18 periph drivers does not include support for alternate pins for I2C1 (P2.18/2.17).  Without being able to specify the alternate pins during init, the MXC_I2C_Recover call which is issued during the initialization process fails, causing the initialization itself to fail.

Current work around is modifying pins_me18.c to specify the alternate pin configuration.

Other I2C instances and peripherals should be checked for similar problems.

Jake-Carter, 2024-03-15:
@BrentK-ADI I think this might be a typo in the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/max32690.pdf) (?).  I2C1A is listed on AF1 as P0.11/P0.12 _and_ P2.17/P2.18 for the 140 WLP...

BrentK-ADI, 2024-03-17:
@Jake-Carter I agree there's a typo/confusing naming in the datasheet.  The same applies for P0.30/0.31 & P2.7/P2.8 (I2C0A) and P0.12/0.13 & P1.7/1.8 (I2C2C).    However, datasheet aside, these are valid alternate pins for the I2C peripherals.  I verified on the APARD board which has the 140WLP.

Also note the [errata](https://www.analog.com/media/en/technical-documentation/data-sheets/max32690_a2_errata_rev2.pdf) regarding these pins for the RevA2 silicon:
[image]

Jake-Carter, 2024-03-18:
I see, thanks.  I wasn't aware of the errata.

I think I'll update the definitions to enable both sets of pins simultaneously

PR #958 (2024-03-21): fix(PeriphDrivers): Fix Unusable MAX32690 I2C Pins 
## Pull Request Template

### Description

Fixes #949 

The ME18 assigns the same alternate function to both sets of I2C pins.  This is logged as an errata (Rev A2 Errata 16), but in order to make the second set of pins usable our drivers will now configure both sets simultaneously.

Doxygen has been updated on the init function to highlight this change:

[image]

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related [... cut]
closed 2024-03-21 by Jake-Carter