analogdevicesinc/msdk #965: Using the ADC SFR Interface to Load the Reference Trim

TerryWoo123, 2024-03-27:
1. I received void MXC_ADC_RefCalibration(mxc_adc_revb_regs_t* adc, mxc_adc_req_t *req) this function from Sanjay.Jaroli last year. this is used to load trim value of ADC reference, to improve accuracy of ADC, for MAX32672-B2 variant.
2. this function works well. with internal 2.048V, we see output result is significantly colser to the  atual voltage.
3. ADC feature in MAX32662 is not fully the same as MAX32672-B2. According to UG of MAX32662, I ported MXC_ADC_RefCalibration(mxc_adc_revb_regs_t* adc, mxc_adc_req_t *req) for MAX32662, but I don't see any slight/significant improvement of ADC accuracy. with 2.048 internal reference, there is about 40mV gap between ADC output and actual voltage.

please help to check why my code for MAX32662 doesn't work well? 
Once we confirm that these code work well, please merge this feature to MSDK in our next release.

thanks
[adc_revb.zip](https://github.com/analogdevicesinc/msdk/files/14770082/adc_revb.zip)

Terry Wu
2024.03.27

TerryWoo123, 2024-03-27:
After a bug fix, it works well with with 2.048V internal reference in MAX32662. Please check new version code.
[adc_revb.zip](https://github.com/analogdevicesinc/msdk/files/14771725/adc_revb.zip)
Please help to verify my code and update it to MSDK.

Jake-Carter, 2024-04-03:
Thanks @TerryWoo123, it should be relatively straightforward to pull this into the MSDK.  There are just a few changes we will need to make:
1. We try to make sure any part-specific code is implemented in the `adc_mexx.c` files.
2. As you noted the bias and wake-up counter values will need to be updated based on the sample rate.  It seems straightforward to implement based on the table in the UG.  The sample-rate calculation actually seems like the most complicated part of this.

My understanding is that Rev A1 of the MAX32672 is the only special case.  Otherwise, the MAX32662 and MAX32672 are identical.  Is that correct?

TerryWoo123, 2024-04-06:
1.  MAX32672-A1 is not on sell any more. We don’t need to pull any code for MAX32672-A1.
  2.  MCR_ADC_CFG3 register is added in MAX32672-B2, but I don’t see this register in MAX32662.

[External]