analogdevicesinc/msdk #976: How to turn on the FPU and DSP part of a Cortex-M

liuyuxivapor, 2024-04-03:
Hi, I am using Eclipse for MAXIM for a proj containing some data processing work on MAX78000. How to the **FPU and DSP switch** of a Cortex-M4 while building and deploying, cause it really works faster. Thanks a lot!

Jake-Carter, 2024-04-03:
Hi @liuyuxivapor

To enable the hardware-accelerated FPU set the `MFLOAT_ABI` build configuration variable to `hard`.
- See [Build Variables for Arm Cortex-M4 Cores](https://analogdevicesinc.github.io/msdk//USERGUIDE/#build-variables-for-arm-cortex-m4-cores) and [Build Configuration Variables](https://analogdevicesinc.github.io/msdk//USERGUIDE/#build-configuration-variables) for more details.

The MSDK also comes with support for the CMSIS-DSP library.  To enable it, set `LIB_CMSIS_DSP = 1`.
- See [Build Variables for Toggling Libraries](https://analogdevicesinc.github.io/msdk//USERGUIDE/#build-variables-for-toggling-libraries) and [CMSIS-DSP](https://analogdevicesinc.github.io/msdk//USERGUIDE/#cmsis-dsp)

For a complete example, the [ARM-DSP examples](https://github.com/analogdevicesinc/msdk/tree/main/Examples/MAX78000/ARM-DSP) are good reference projects.

Jake-Carter, 2024-04-04:
Will close this out - let me know if we can clarify anything else.

closed 2024-04-04 by Jake-Carter

liuyuxivapor, 2024-04-05:
Thanks a lot. I have finished FPU configuration with this, but according to the user guide of Cortex-m4 and cmsis-dsp, a dsp hardware part is also embeded in the m4 core of max78000, so is it possible to use the dsp part in configuration? I did not see instructions about this in msdk user guide.

Jake-Carter, 2024-04-16:
@liuyuxivapor sorry for the delay, I've been at a trade show for the past week with limited development access.

Perhaps you can share with me the exact document/wording you're looking at?  In general, the Cortex M4 has a DSP-focused _instruction_ set that the CMSIS-DSP library uses to accelerate its performance.  I've attached an Arm whitepaper that goes into more detail.

[7563.ARM white paper - DSP capabilities of Cortex-M4 and Cortex-M7.pdf](https://github.com/analogdevicesinc/msdk/files/14999044/7563.ARM.white.paper.-.DSP.capabilities.of.Cortex-M4.and.Cortex-M7.pdf)

But in general the M4 core itself does not have a true DSP block.  Usually it is an external block to the CPU that's added as a peripheral.  Or in the case of the MAX78000, we have the CNN accelerator that can perform similar tasks using neural networks.