analogdevicesinc/msdk #984: How to Unlock a MAX78000 That Can No Longer Be Programmed

PYL4869, 2024-04-09:
Hi
for a custom board design, I had some problems with my DAPLINK drive, you can refer to this[https://github.com/ARMmbed/DAPLink/issues/1070](url), It can download normally, but erase.act cannot be used, about this question[https://github.com/MaximIntegratedAI/MaximAI_Documentation/blob/main/MAX78000_Feather/README.md#how-to-unlock-a-max78000-that-can-no-longer-be-programmed](url), I may not be able to use "erase.act"allowing the device to be programmed again, Is there any way to unlock the non-programmable MAX78000?

Besides that, how is your "UART 0" related to "printf"? This is the only one I found in the startup folder, and it doesn't seem to associate "UART 0" with "printf".

[image]

EricB-ADI, 2024-04-12:
@Jake-Carter May be able to help you more with your first question. 

Regarding UART0 and printf. The board.h file refers to ```UART0``` as ```CONSOLE_UART```. 
Then Inside the MiscDrivers folder is the stdio.c file, which implements 
the stubs required for printf. You can see it defining the UART from ```CONSOLE_UART`` at 
https://github.com/analogdevicesinc/msdk/blob/b71451d3b8ae85c88a6e417b9983dbc792609fbf/Libraries/MiscDrivers/stdio.c#L79

Then the putc implementation which printf requires uses our normal UART drivers

https://github.com/analogdevicesinc/msdk/blob/b71451d3b8ae85c88a6e417b9983dbc792609fbf/Libraries/MiscDrivers/stdio.c#L188

Jake-Carter, 2024-04-16:
@PYL4869 see #883

You may be encountering a bug I found with DAPLINK firmware SHA "a209", which caused it not to drive the RSTN pin correctly.

PYL4869, 2024-04-17:
@Jake-Carter 
Hi, I may not have made this clear.
for a custom board design, I had some problems with my DAPLINK drive, I made my DAPLink device using "STM32103CBT6", After my tests, I found that I could not erase MAX78000 using "eraser.act" in "MSD command", you can see [this](https://github.com/ARMmbed/DAPLink/issues/1070),  I may not be able to use "erase.act"allowing the device to be programmed again, Is there any other way to unlock the unprogrammable MAX78000?

For [this](https://github.com/analogdevicesinc/msdk/issues/883#issuecomment-1905184961), Either in the MAX78000FTHR or my custom board design, when the download cannot be performedthis VS Code "erase flash" build task does not work.

[image]

[image]

For [this](https://github.com/analogdevicesinc/msdk/issues/883#issuecomment-1928584046), because I use STM32103CBT6 to make my DAPLink device, it should not solve my problem.

For my custom board design, how to Unlock a MAX78000 That Can No Longer Be Programmed ?

Jake-Carter, 2024-04-17:
@PYL4869 since you are using an STM device as your DAPLINK device I can't provide much support on the implementation.  I would suggest reaching out to ST support instead.

Can you at least connect to the MAX78000FTHR at all?  Try the following command.  It will print some extra debug info as well (`-d3`)

```shell
openocd -s $MAXIM_PATH/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/max78000.cfg -c "init;" -d3
```

(Note the `MAXIM_PATH` environment variable should be set, or replace it with the full path to the MSDK install)

If it also fails, then you may have some fundamental issues with the DAPLINK firmware/build.  The errors above look like the debugger cannot even find the MAX78000 at all.  So you will need to resolve that before you can proceed to erasing the device

closed 2024-05-02 by PYL4869

jedddi, 2025-06-14:
Hello @PYL4869 , having the same right now. Have you ever resolved the issue?

PYL4869, 2025-06-15:
@jedddi  Hello,I'm not sure which one you're talking about. 
Can you describe it in detail?

jedddi, 2025-06-15:
@PYL4869 Im having the same issues trying to flash to the MAX78000 and this is how exactly it looked like. Was wondering if you we are able to fix this issue. 

[image]