analogdevicesinc/msdk #1002: Can't create new "Analog Devices Microcontroller" project

marcrubenstein, 2024-04-25:
Creating new "Analog Devices Microcontroller" fails with null pointer exception. Ubuntu Jammy, MSDK version SFW0018720B

```
Unhandled event loop exception

java.lang.NullPointerException: Cannot read the array length because the return value of "java.io.File.listFiles()" is null
	at maximIntegratedProjects.OptionsPage.createControl(OptionsPage.java:72)
	at org.eclipse.jface.wizard.Wizard.createPageControls(Wizard.java:178)
	at org.eclipse.jface.wizard.WizardDialog.createPageControls(WizardDialog.java:744)
	at org.eclipse.jface.wizard.WizardDialog.createContents(WizardDialog.java:636)
	at org.eclipse.jface.window.Window.create(Window.java:431)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1094)
	at org.eclipse.ui.internal.actions.NewWizardShortcutAction.run(NewWizardShortcutAction.java:119)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:414)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5855)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1529)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:5065)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4517)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:643)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:550)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:171)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:402)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1440)

eclipse.buildId=4.26.0.20221201-1200
java.version=17.0.5
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

```

Jake-Carter, 2024-04-29:
Thanks @marcrubenstein, I've been able to replicate some issues with the new project wizard as well.  On my machine I don't get the exception message, but nothing opens when I attempt to create the project.

Tagging @lorne-maxim as well

Jake-Carter, 2024-04-29:
In the meantime, you can [import examples](https://analogdevicesinc.github.io/msdk//USERGUIDE/#importing-examples) as a workaround

marcrubenstein, 2024-04-29:
@Jake-Carter Thank you. Originally, I did not get an exception message (as you said, nothing happened). However, when I enabled the "Error Log" view in eclipse, this error message was there

I'll use the workaround in the meantime. Thanks!

Jake-Carter, 2024-10-03:
FYI this was an issue with the `setenv.sh` scripts for Linux/MacOS.  The logic for auto-settings MAXIM_PATH was failing in some edge cases between bash/zsh.  It has been fixed and is just waiting to go out in the next release.

Fixed file attached.

[setenv.zip](https://github.com/user-attachments/files/17250955/setenv.zip)

Jake-Carter, 2024-11-05:
Should be fixed as of the [Oct 2024 Release](https://github.com/analogdevicesinc/msdk/releases/tag/v2024_10)

closed 2024-11-05 by Jake-Carter