analogdevicesinc/msdk #1003: MAX32660 EVSYS Board Can't Flash

yuval5295, 2024-04-28:
Hey, 

I was running Hello_World example on it But I can't flash my board, it is giving the error:
**Remote communication error. Target disconnected.:**

I am using VS (Eclipse made me too much troubles) and unfortunately I had never succeed to flash it.
I did all the instructions step by step - updated the adapter firmware and erase but still no success.

this is my details on DAPALINK:
Unique ID: 04211702bfb8038600000000000000000000000097969906
HIC ID: 97969906
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Incompatible image detection: 0
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0256
Git SHA: 34fe95d79afeba62ef097774ecc609ef8c2b74ea
Local Mods: 1
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0xd7617efe
Remount count: 0
URL: http://www.maximintegrated.com/max32660

this is what written on terminal:
[image]

looking forward for any help :)

Jake-Carter, 2024-04-29:
Hi @yuval5295, have you ever been able to program the board?  Or this is the error you're seeing on a first-time attempt?

yuval5295, 2024-04-29:
Hey Jake!
I have never succeeded to program the board. I just started to try run the
first example and it’s keep fail flashing.

Jake-Carter, 2024-04-29:
Do you have a jumper installed on JP1 (labelled "Target VDD")?  

You'll need to select either 3.3V or 1.8V.  Either setting should work, but you'll need to select at least one to supply power to the MAX32660

yuval5295, 2024-04-29:
Yes I have and I put the jumper on 1.8V option.

Jake-Carter, 2024-04-29:
We re-tested with a board on our side and that latest PICO firmware you're using should work.

Can you run the following command from your VS Code terminal, and paste the output?  It will print some additional debug info.

```shell
openocd -s $MAXIM_PATH/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/max32660.cfg -c "init;reset halt" -d3
```

The error you're getting is the generic error when there is a connection issue to the micro on the SWD port, the micro is powered off, the host PC can't find the debugger, etc.

Jake-Carter, 2024-05-04:
@yuval5295 looks like the image didn't make it to the ticket from the email.  Can you paste in the terminal output, or attach the image to the ticket on Github?

yuval5295, 2024-05-06:
Sure, here it is. [image: image.png]

Jake-Carter, 2024-05-09:
Sorry, still not showing:

[image]

Can you paste the actual terminal output?  `CTRL+SHIFT+C` should let you copy from the VS Code terminal

Jake-Carter, 2024-05-09:
Alternatively you can run

```shell
openocd -s $MAXIM_PATH/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/max32660.cfg -c "init;reset halt" -d3 > openocd_log.txt 2>&1
```

and attach "openocd_log.txt"

yuval5295, 2024-05-11:
cfg -c "init;reset halt" -d3
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0
(2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to ***@***.***>
User : 3 3 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 7 options.c:52 configuration_output_handler():
Debug: 5 8 options.c:233 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 10 options.c:234 add_default_dirs():
pkgdatadir=/usr/local/share/openocd
Debug: 7 13 options.c:235 add_default_dirs():
exepath=C:/MaximSDK/Tools/OpenOCD
Debug: 8 16 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 18 configuration.c:35 add_script_search_dir(): adding
C:/Users/97254/AppData/Roaming/OpenOCD
Debug: 10 21 configuration.c:35 add_script_search_dir(): adding
C:/MaximSDK/Tools/OpenOCD/../share/openocd/site
Debug: 11 25 configuration.c:35 add_script_search_dir(): adding
C:/MaximSDK/Tools/OpenOCD/../share/openocd/scripts
Debug: 12 29 command.c:155 script_debug(): command - ocd_find
interface/cmsis-dap.cfg
User : 13 32 command.c:608 command_run_line(): embedded:startup.tcl:28:
Error: Can't find interface/cmsis-dap.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 28

yuval5295, 2024-06-01:
> @yuval5295 <https://github.com/yuval5295> are you on Windows? Sorry - I
> used the Linux syntax for environment variables. On Windows it should be:
> (%MAXIM_PATH% instead of $MAXIM_PATH)

Hellp Jack!
it seems nothing printed on the command line after I sent this command but
it makes D1 blink green light fast (which not before I send it).
What should I do next?

Thanks!
Yuval.

Jake-Carter, 2024-06-04:
Hi @yuval5295,

That actually means that the command has successfully connected to the MAX32660.  The OpenOCD server has launched and is waiting for a GDB client connection.  Essentially it means that the debugger is working properly with the command.

This means that there could be some problem specific to the VS Code setup.

Have you set `MAXIM_PATH` correctly in your user settings.json file?  See [here](https://analogdevicesinc.github.io/msdk//USERGUIDE/#getting-started-with-visual-studio-code)

Please copy/paste the entire contents of your settings.json file here as well.

yuval5295, 2024-06-09:
Hello Jake!

Great news, you are totally right. the problem was the MAXIM_PATH I set, by
accident I put two // instead of one!
Thank you so much for your help and I'm sorry to bother you with such
stupid mistake.

Also I wanted to ask, can you tell me what should I check if I can't see
the prints on VS terminal? I used PuTTy instead but it seems that VS
Terminal should work as well.

Thank you again,
Yuval.

Jake-Carter, 2024-06-11:
Hey, no problem at all.

You can use the [Serial Monitor extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor).  This is what I use.

Just be aware it has a small quirk - it will send a special serial command to reset the micro when you connect to the serial port...  This can make debugging a bit strange if you're not aware of it.  It's why I have not officially recommended it in the User Guide.  But the extension otherwise works very well.

closed 2024-06-11 by Jake-Carter