analogdevicesinc/msdk #1009: CSI2 retention problem in standby mode

isztldav, 2024-05-08:
**Board**: devkit max78002

**In short**:
I took the provided [CSI2 example](https://github.com/analogdevicesinc/msdk/tree/main/Examples/MAX78002/CSI2), implemented "standby mode", and added the call to `int mipi_camera_sleep(int sleep)`.  
While the wake/sleep cycle seems to work well, the camera capture will throw errors and after some time suddenly work (for one frame).

**Source code**:
I just changed the main file: [main.zip](https://github.com/analogdevicesinc/msdk/files/15245393/main.zip)  
basically adding `static void _go_to_sleep(void)` and `static void _setup_btn_wake(void)`.

**Image**:
[image]

Thank you for your help!

Jake-Carter, 2024-05-13:
Thanks @isztldav - I've been able to replicate the issue.  This is reminiscent of some deep-rooted timing issues in our CSI hardware I've dealt with in the past.  I hadn't tested captures from STANDBY.

There also seems to be some issues with the sleep functions specific to the OV5640 drivers.  If I remove the `mipi_camera_sleep` calls I can get things working in Sleep Mode and Low Power Mode, which leave the peripherals fully powered on.  think we need to re-examine our sleep mode functions for the camera too.

Micro-Power Mode (UPM) fails with the same random CTRL error flags.  That mode actually puts the peripherals in state retention, so there seems to be some core issue with the state retention logic for the CSI2 peripheral in general.

isztldav, 2024-05-22:
Thank you for confirming this. For now my best workaround is to spam the capture function until magically one frame is transmitted from the camera.