analogdevicesinc/msdk #1037: Requesting assistance with an error while running a demo in max78000 fthr board.

maanasi8, 2024-06-06:
```
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7) OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

This is the error I am facing while debugging and running the demo hello world in max78000 fthr board.
I am getting this output in my terminal of eclipse MaximSDK.
I have folllowed all the instructions specifies in the official github page of maxim and their user guide.
The build is successful and it produceses a .elf file then when i debug it it stops at the above when it should listen and work properly and give me a output.

Jake-Carter, 2024-06-12:
Hello @maanasi8, that output is indicating that the OpenOCD server is launched and waiting for a GDB client to connect to it.

Can you capture a full screenshot of your Eclipse window when you launch the debug session?

Jake-Carter, 2024-07-01:
@maanasi8 closing for inactivity.  Will re-open if necessary

closed 2024-07-01 by Jake-Carter