analogdevicesinc/msdk #1040: MAX78000 - Inconsistency between User Guide and SDK

alessandro-montanari, 2024-06-10:
Hello,

I'm working on low power modes and wakeup on MAX78000 and I noticed that the user guide says to write 1 to the lpwkst_clr field on register `PWRSEQ_LPCN` to clear the wakeup status flags (`PWRSEQ_LPWKST0`, `PWRSEQ_LPWKST1`, ...).
However, in the SDK (Libraries/PeriphDrivers/Source/LP/lp_ai85.c) the function (`MXC_LP_ClearWakeStatus()`) clears them by writing all 1s into the registers directly.

I tried to modify the function with this line `MXC_PWRSEQ->lpcn |= MXC_F_PWRSEQ_LPCN_LPWKST_CLR;` but it seems to have bricked my device. I couldn't reset it or re-flash it. I had to remove power to it and then it came back.

What is the best way to clear those registers?
Thanks

Jake-Carter, 2024-06-12:
Hi @alessandro-montanari, I will check with our hardware designers and internal tickets to see if there are any known issues with that bit.  

At the moment I would say continue to use our existing implementation that writes to the fields directly.  This may be a workaround to some issue.

I will let you know what I find

Jake-Carter, 2024-11-05:
@alessandro-montanari I know it has been some time on this issue...  apologies for the delay.  I have actually been in the process of transitioning out of ADI and am cleaning up some old tickets.

I was not able to find any internal documentation on any known issues.  I was also able to lightly test `MXC_PWRSEQ->lpcn |= MXC_F_PWRSEQ_LPCN_LPWKST_CLR;` and was able to at least write to the bit without bricking the device.  It seems like it should be operating normally, and that clearing the fields manually may just be an oversight when writing the drivers.  The authors may not have known about the bit.

If you can attach any sample code/details for a program that can consistently brick the device on this bit that would be very helpful.

closed 2024-11-05 by Jake-Carter

alessandro-montanari, 2024-11-29:
Thanks for checking this @Jake-Carter.
In the end, we didn't need to read the wakeup status flags, so we didn't investigate this further.
If we go back to test this again `MXC_PWRSEQ->lpcn |= MXC_F_PWRSEQ_LPCN_LPWKST_CLR;` I'll post here.