analogdevicesinc/msdk #1044: FatFS using SDHC is hardcoded to 1-bit mode in diskio.c

JordanAceto, 2024-06-11:
Hi there,

I noticed slow SD card writes when using the SDHC library. Probing the data lines with an oscilloscope revealed that only the `DATA0`, `CLK`, and `CMD` lines were active. The `DAT1` .. `DAT3` lines were permanently high during data writes.

Browsing through the code and using the debugger, I found that the data width is hardcoded with the enumeration `MXC_SDHC_LIB_SINGLE_DATA` in diskio.c. The relevant line for the write function is shown below. The matching read function has the same hardcoded enum. I am using FF15.

https://github.com/analogdevicesinc/msdk/blob/5955fcf103fdb42f5a424dcc08eedbb942b520b1/Libraries/SDHC/ff15/source/diskio.c#L199

After manually changing the hardcoded enums to `MXC_SDHC_LIB_QUAD_DATA` in diskio.c, I do see all 4 data lines wiggling, and the SD card write speeds increase as expected.

Is there a reason for the SDHC to be hardcoded to 1-bit mode when using FatFS/can the ability to use all 4 data lines be added to the MSDK?

Thanks!

EricB-ADI, 2024-06-11:
What chip are you using? I do not know the full history, but not all of our chips support QSPI operation, for ease it was most likely set to that to not have to deal part to part variation. I will add this to my list and will update this sometime soon.

JordanAceto, 2024-06-11:
We are using MAX32666. The current code is on a FTHR board, but we will eventually move it to a custom PCB. Thanks!

Jake-Carter, 2024-06-11:
Thanks @JordanAceto, this is just a legacy hard-coded config.  We can and should expose this as a config option to the API.

Jake-Carter, 2024-06-11:
Ah looks like @EricB-ADI is already on it in #1045

closed 2024-06-25 by Jake-Carter

naveedakhter01, 2025-06-19:
> What chip are you using? I do not know the full history, but not all of our chips support QSPI operation, for ease it was most likely set to that to not have to deal part to part variation. I will add this to my list and will update this sometime soon.

Does Max78000 support 4-bit SD Card mode?