analogdevicesinc/msdk #1074: MAX32660: SPI v2

trevorcmit, 2024-07-08:
Hello msdk developers,

I have a quick questions about the MAX32660 and the usage of the new SPI v2 SDK. Is the v2 meant to be targetable for any of the MAX326xx chips? Or is it specifically for MAX7xxxx series? I have tried following the MSDK User Guide about updating the build configuration variables and that does not seem to change my code to V2. Is there anything else that must additionally be done to utilize the SPI v2? I have verified my MSDK is updated to the newest release and can see the v2 code files in the PeriphDrivers in the MaximSDK folder location.

Jake-Carter, 2024-07-16:
Hi @trevorcmit,

SPI v2 isn't fully ported to all the micros yet.  I thought we had added the current status to the developer note already but I guess not.

It needs a somewhat non-trivial implementation file to be created for each micro.  (see [spi_ai87_v2.c](https://github.com/analogdevicesinc/msdk/blob/main/Libraries/PeriphDrivers/Source/SPI/spi_ai87_v2.c) vs [spi_ai87.c](https://github.com/analogdevicesinc/msdk/blob/main/Libraries/PeriphDrivers/Source/SPI/spi_ai87.c))

@sihyung-maxim can you update the UG with the list of SPI v2 supported parts?

trevorcmit, 2024-07-18:
Thanks for the response @Jake-Carter. It does seem very non-trivial; is the plan for all the micros to eventually be implemented?

Jake-Carter, 2024-07-19:
Yes, the plan is to eventually migrate them all.  We're working with a very small team atm so it may take some time.  First we need to catch up on our backlog.  Contributions are very much welcome :)