analogdevicesinc/msdk #1082: MAX32690 Error writing to flash

HfTFranz, 2024-07-17:
Hello MSDK Devs, 

I am fellow ADI employee attempting to use the MAX32690 for an internal project.
I am not sure how or why, but in the middle of a program that uses the microcontroller via USB CDC-ACM, the microcontroller stopped responding.
Now I am unable to flash to it.

Console log:
```
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.0
Info : CMSIS-DAP: Serial# = 0426170224bd030a00000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : [max32xxx.cpu] Cortex-M4 r0p1 processor detected
Info : [max32xxx.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for max32xxx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : [max32xxx.cpu] target was in unknown state when halt was requested
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory at 0x00004340
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory at 0x0000433e
Info : [max32xxx.cpu] external reset detected
[max32xxx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x0000433c msp: 0x2000a8e0
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000fff4 msp: 0x2000a900
Warn : No RTOS could be auto-detected!
Warn : No RTOS could be auto-detected!
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0x00000222 msp: 0x2000a8e0
Error: timed out while waiting for target halted
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x10000000 at offset 0x00000000
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000fff4 msp: 0x2000a900
Info : dropped 'gdb' connection
shutdown command invoked

```

I have tried doing both erase flash and copy pasting erase.act to the DAPLINK with no avail.
Do you have some ideas as to how we could possibly fix this without replacing the unit? Or is this unfortunately unfixable?

Thanks, Franz

HfTFranz, 2024-07-17:
Here is the log when I tried to execute erase flash. I noticed that it also has an error when writing to this memory address: 0x4002900c

```
 *  Executing task: openocd -s C:/MaximSDK/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/max32690.cfg -c "init; reset halt; max32xxx mass_erase 0;" -c exit 

Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.0
Info : CMSIS-DAP: Serial# = 0426170224bd030a00000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz      
Info : SWD DPIDR 0x2ba01477
Info : [max32xxx.cpu] Cortex-M4 r0p1 processor detected
Info : [max32xxx.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for max32xxx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Warn : [max32xxx.cpu] target was in unknown state when halt was requested
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000fff4 msp: 0x2000a900
Info : SWD DPIDR 0x2ba01477
Error: Failed to write memory at 0x4002900c
max32xxx mass erase complete

Info : [max32xxx.cpu] external reset detected
 *  Terminal will be reused by tasks, press any key to close it. 
```

Jake-Carter, 2024-07-19:
Hi @HfTFranz

The problematic address falls within the FLC0 instance:

[image]

and seems to be the Flash Busy Flag bit.  I suspect OpenOCD is hitting the error condition mentioned in the note, and the pending flag is set.

[image]

Can you attach the project (and binary if you have it) for the program that caused the failure?  This has the symptoms of some sort of flash controller failure but I'd like to take a look at the application.

HfTFranz, 2024-07-24:
Hello @Jake-Carter, 

Not sure if I can attach the project here. However, flashing anything (even the examples) would not work. I really think the microcontroller I got just failed.
I just got a new microcontroller IC and the program now flashes and runs just fine.

closed 2024-07-24 by HfTFranz