analogdevicesinc/msdk #1121: Build Examples Is very slow

EricB-ADI, 2024-08-19:
Can we add some more logic so it we just change a single file it doesn't build every single example for every single chip? Most changes are not effecting multiple parts.

Jake-Carter, 2024-08-23:
I've started a solution on https://github.com/Jake-Carter/msdk/tree/feat/build_test_optimizations

That workflow is nice.  We can clean things up with it.

[image]

This leverages the fact that the target micro name is in the filepath.  Source folders, `meXX` files, and `rev_XX` files are more problematic.

The hardest problem is that there is no dependency mapping between libraries and target micros.  For example, if a file is changed in MAXUSB which examples should be re-tested?  If we change tmr_reva.c which examples are affected?

I want to avoid hard-coding a bunch of lists.  Open to suggestions.

What we have now is slow, but it auto-detects new examples and catches everything

EricB-ADI, 2024-08-23:
Kind of a multilevel issue.
A) If we don't know the dependency fall back and build all. We can probably leverage jobs in GitHub to run the subsections in parallel. So something like build AI chips., build BLE, etc. in different jobs so that we get the parallelism of GitHub on top of whatever the script is already doing.  
B) We create a dependency list for the build to use

PR #1135 (2024-09-26): feat(Build): Optimize Build Examples Test
### Description

Closes #1121 

The `.github/workflows/scripts/build.py` script now accepts a `--change_files` optional argument.

```shell
  --change_file CHANGE_FILE
                        (Optional) Pass a text file containing a list of space-separated or new-line separated changed files. The build
                        script will intelligently adjust which parts it tests based on this list.
```

It takes about 40s to generate the dependency map on the Github runners.

Now, if a change is made to Cordio only the BLE micros will be re-tested.  If [... cut]
closed 2024-09-26 by EricB-ADI