analogdevicesinc/msdk #1125: Import Example Project on MAX78002

smtalds, 2024-08-21:
Hi everyone,
There is a problem about import the MAX78002 example projects. When i want to add a example project like CNN project, there is not listed any project.
[image]

So when i import project with **Existing Projects into Workspace**, includes is missing. I can not find solution.
[image]

When i checkout [examples list of MAX78002](https://github.com/analogdevicesinc/msdk/blob/main/Libraries/Boards/MAX78002/EvKit_V1/examples.txt), CNN examples is missmatch. Can you check out please ? When i add the project in the list myself, build folders not listed on project explorer. Thanks for answering.

smtalds, 2024-08-21:
When i import two project like` facial_recognition` and `pascalvoc-retinanetv7_3`, include files seems like
[image]
 So eclipse not call all libraries, so i want to change something, i can not find solution. I dont want to edit sub folders. Maybe i may ruin it.
 @Jake-Carter Can you help me please :)

Jake-Carter, 2024-08-27:
Thanks @smtalds - looks like my project generator was missing detection of some header files in nested folders.  Just updated the project in the PR above.

On the new project wizard - we'll need to investigate why the CNN examples aren't showing up in the list...  @sihyung-maxim  any ideas?

sihyung-maxim, 2024-08-27:
The new project wizard grabs the list of a part's examples from `$(MAXIM_PATH)/Libraries/Boards/{TARGET}/{BOARD}/examples.txt` file and then verifies that those directories exist in your MSDK install. If not, then they don't show up in the dropdown options.

Looks like the MAX78002 examples list was not updated during the last release. I've opened a PR to update it. If you would like to create the change locally in your official MSDK install, check PR #1139.

Jake-Carter, 2024-08-28:
@smtalds PRs merged into main.

CNN examples should show up in the new project wizard, and include paths for `pascalvoc-retinanetv7_3` should be fixed.

PR #1138 (2024-08-28): fix(Other): Update pascalvoc-retinanetv7_3 Eclipse Include Paths
## Pull Request Template

### Description

Fixes #1125 

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related GitHub issues [using a keyword](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword)
- [ ] (Optional) Provide info on any relevant functional testing/validation.  For API changes or significant fea [... cut]