analogdevicesinc/msdk #1161: Hardcoded pin definitions in `pins_me1x.c` are inconvenient for some workflows

JordanAceto, 2024-09-09:
The hardcoded pin definitions in `msdk/Libraries/PeriphDrivers/Source/SYS/pins_me1x.c` cause issues when not using all the possible pins for a specific peripheral, like SPI busses.

For example, we are using a MAX32666 with a basic 4-wire SPI bus on SPI1 (CS, SCK, MOSI/SDIO0, and MISO/SDIO1 only). When we call `MXC_SPI_Init(...)` it initializes all the pins we want, **but** it also initializes the P0.20/SDIO2 and P0.21/SDIO3 pins as function AF2, which we don't want.

The SPI init call is here: https://github.com/analogdevicesinc/msdk/blob/79af4e9980b6141f2cb9797e6085727e88d7fee3/Libraries/PeriphDrivers/Source/SPI/spi_me14.c#L95
And the pin definitions are here: https://github.com/analogdevicesinc/msdk/blob/79af4e9980b6141f2cb9797e6085727e88d7fee3/Libraries/PeriphDrivers/Source/SYS/pins_me14.c#L92

We are using pins P0.20 and P0.21 for other purposes, and we don’t want their config to be overwritten by initializing an unrelated SPI bus. Since `gpio_cfg_spi1` is `const` we can't easily just edit the fields of the `mxc_gpio_cfg_t`.

The same thing comes up with the SDHC peripheral, we are using the Write Protect pin P1.6 for something else, but any time we init the SDHC bus it undoes our desired pin config.

This is an issue because we are dynamically turning peripherals on and off and multiplexing some pins, so it gets unwieldy to ensure that we always re-init the pins for an unrelated peripheral when we call a SPI/SDHC init function.

Does this make sense, and is there any guidance on how to manage this?

My first instinct is to create a custom `pins_me14.c` file that we compile and link instead of the one in the MSDK SYS dir. Is there an easy way we can use `project.mk` to disable the  `msdk/Libraries/PeriphDrivers/Source/SYS/pins_me14.c` file from being compiled and linked, and instead sub in our own custom version? Other suggestions are welcome.

I have been messing around with the build system a bit, but I am not sure what is the best way to accomplish the above. We'd rather not make everyone on the team edit the `msdk/Libraries/PeriphDrivers/Source/SYS/pins_me14.c` file that comes with the installation.

TL;DR how can we stop calls to `MXC_SPI_Init(...)` from configuring all the QSPI pins when we only want 3-or-4-wire SPI?

Thanks!

Jake-Carter, 2024-09-09:
Hi @JordanAceto,

One option you have is to define `MSDK_NO_GPIO_CLK_INIT` in your project.  We added this to support Zephyr, which has its own GPIO and clock configuration system.  If this is defined, the GPIOs will not be configured during peripheral `Init` routines and you can manage them directly.

```Makefile
# project.mk
PROJ_CFLAGS += -DMSDK_NO_GPIO_CLK_INIT
```

See here for SPI: https://github.com/analogdevicesinc/msdk/blob/79af4e9980b6141f2cb9797e6085727e88d7fee3/Libraries/PeriphDrivers/Source/SPI/spi_me14.c#L51

Looks like we're missing it for SDHC though...  Will need to add that in.  https://github.com/analogdevicesinc/msdk/blob/79af4e9980b6141f2cb9797e6085727e88d7fee3/Libraries/PeriphDrivers/Source/SDHC/sdhc_me14.c#L47

Jake-Carter, 2024-09-09:
> My first instinct is to create a custom pins_me14.c file that we compile and link instead of the one in the MSDK SYS dir. Is there an easy way we can use project.mk to disable the [...]

This is an interesting option and would help avoid cluttering up your application's setup phase with `MXC_GPIO_Config` calls.

There's no way currently, but I think it could be implemented relatively easily as a `PINS_FILE` option.  I'll add this as an enhancement option since I think it'd also be useful for users who want to change VDDIO -> VDDIOH, etc.

Jake-Carter, 2024-09-09:
Np!  One other thing to note is that the clock enable is also skipped.  (UART for example:  https://github.com/analogdevicesinc/msdk/blob/79af4e9980b6141f2cb9797e6085727e88d7fee3/Libraries/PeriphDrivers/Source/UART/uart_me14.c#L82))

So you'll need to add in a `MXC_SYS_ClockEnable` for the peripherals as well.

Will leave this open to resolve:
- [ ] : Add `MSDK_NO_GPIO_CLK_INIT` to SDHC Init
- [ ] : Add some documentation on this to the UG
- [ ] : Implement `PINS_FILE`

PR #1210 (2024-10-08): feat(PeriphDrivers, Documentation): Add Ability to Override Default Pin Definitions, Document `MSDK_NO_GPIO_CLK_INIT`, Check for `MSDK_NO_GPIO_CLK_INIT` in SDHC Drivers
### Description

Closes #1161 

- Add `MSDK_NO_GPIO_CLK_INIT` to SDHC Init routines.
- Add some documentation on `MSDK_NO_GPIO_INIT` to UG
- Add a `PINS_FILE` build config option for overriding the default pin definitions file, and document in PeriphDriver-specific section of UG

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
-  [... cut]
closed 2024-10-08 by Jake-Carter