analogdevicesinc/msdk #1204: Transmit and Receive letters in GCR in svd

alexkupri-orion, 2024-10-02:
Place: svd files (potentially, dependent files)
Submodule: GCR
Chips: all chips
Registers: 0x74 and 0x78
Please check registers 0x74 and 0x78 in the module GCR for all chips.

Particularly, check all fields of those registers for transmit and receive functions.

One example is that field "LDORF Bypass Enable Delay" is at the bit 14 in the chip max32690, and the field "LDO BB Bypass Enable Delay" is bit 15. In the chip max32657 these fields are exchanged, 15 and 14.
https://github.com/analogdevicesinc/msdk/blob/aee8e41e31f0c2642c3e526aec5e849516f90e5f/Libraries/CMSIS/Device/Maxim/MAX32657/Include/max32657.svd#L2850
https://github.com/analogdevicesinc/msdk/blob/aee8e41e31f0c2642c3e526aec5e849516f90e5f/Libraries/CMSIS/Device/Maxim/MAX32690/Include/max32690.svd#L7454
Theoretically possible, but very-very suspiciously.

Another example is chip max32665.svd, GCR, register 0x74, field at bit 15. Name is "LDORXBYPENENDLY" with "R" for receive, while description "LDOTX Bypass Enable Delay" with "T" for transmit 
https://github.com/analogdevicesinc/msdk/blob/aee8e41e31f0c2642c3e526aec5e849516f90e5f/Libraries/CMSIS/Device/Maxim/MAX32665/Include/max32665.svd#L4344

alexkupri-orion, 2024-10-02:
P.S. Note, I gave only some examples, most likely there are more transmit-receive errors in the given registers, so it is highly recommended to check the GCR registers 0x74 and 0x78 for all chips.

EricB-ADI, 2024-10-02:
Hi there,

MAX32690 is correct. I am working on verifying MAX32657. The base for this chip however is not the MAX32690 so It isn't out of the realm of possibilities they are different. 

For the MAX32665. That is a typo in the description, and I will fix that.

closed 2024-10-07 by EricB-ADI