analogdevicesinc/msdk #1222: Redefine error when using C++

EricB-ADI, 2024-10-10:
I am working on an internal project with c++.

I am getting a redefinition error 
```bash
In file included from /home/eric-busch/Workspace/msdk/Libraries/CMSIS/Device/Maxim/MAX32655/Include/max32655.h:257,
                 from /home/eric-busch/Workspace/msdk/Libraries/PeriphDrivers/Include/MAX32655/mxc_device.h:24,
                 from ../../../../../Libraries/RF-PHY/MAX32655/include/dbb_registers.h:51,
                 from inc/tester.hpp:14,
                 from src/tester.cpp:7:

core_cm4.h:222: warning: "__I" redefined
```

where it is previously defined
```bash
gpio_regs.h:53: note: this is the location of the previous definition
```

core_cm4.h
```c
#ifdef __cplusplus
  #define   __I     volatile             /*!< Defines 'read only' permissions */
#else
  #define   __I     volatile const       /*!< Defines 'read only' permissions */
#endif
#define     __O     volatile             /*!< Defines 'write only' permissions */
#define     __IO    volatile             /*!< Defines 'read / write' permissions */

/* following defines should be used for structure members */
#define     __IM     volatile const      /*! Defines 'read only' structure member permissions */
#define     __OM     volatile            /*! Defines 'write only' structure member permissions */
#define     __IOM    volatile            /*! Defines 'read / write' structure member permissions */
```

gpio_regs.h
```c
#ifndef __IO
#define __IO volatile
#endif
#ifndef __I
#define __I  volatile const
#endif
#ifndef __O
#define __O  volatile
#endif
#ifndef __R
#define __R  volatile const
#endif
```
There must be some place core_cm4.h is getting included after gpio_regs.h
Only reason it is not an issue before is the defs are the same. 

The easiest fix I can do right now is to basically copy the core_cm4.h files. 

If you have a better idea let me know. 

@sihyung-maxim @Jake-Carter

Jake-Carter, 2024-10-10:
@EricB-ADI can you attach that `tester.hpp` file?  Having a hard time replicating

EricB-ADI, 2024-10-11:
@ttmut Linked this https://developer.arm.com/documentation/ka002980/latest

Yes, it should be updated as const will get optimized volatile or not. 
The register script is in an internal repo, so I can play with it there.

EricB-ADI, 2024-10-11:
Addressed in #1228

closed 2024-10-11 by EricB-ADI