analogdevicesinc/msdk #1248: BLE_FreeRTOS example does not work in tickless mode on MAX32666

sullivanmj, 2024-11-01:
I am attempting to run the BLE_FreeRTOS example for MAX32666. When i do not have tickless configured, I am able to see the DATS peripheral's advertisements and connect to it from a smartphone BLE explorer app.

Once I configure tickless and run the example, I notice the following:

0. I see that the power consumption has dropped to a super low level - this is great, and exactly what I was hoping to see.
1. I can see advertisements, but the signal strength has dropped by about 55 dBm (on a MAX32666FTHR from about -35 to -90, and on a MAX32666EVKIT with a known good Bluetooth antenna connect to J1, it falls from about -45 to -100)
2. When I attempt to connect with the same smartphone BLE explorer app, the connection cannot be established.

EricB-ADI, 2024-11-04:
I am investigating this now. It seems to be an issue with the interaction of the physical layer and sleep.

EricB-ADI, 2024-11-04:
Ok, ``int freertos_permit_tickless(void)`` function does not check whether or not the radio is active. So, the part wakes up starts sending, but before it finishes the scheduler puts it back to sleep. The MAX32665/MAX32666 has no implementation to check if the radio is enabled. I have created the implementation and need to go through the proper steps to add it and verify the sleep is still working.

sullivanmj, 2024-11-08:
@EricB-ADI Thanks for the work on this one. I confirmed that I am now able to connect with a phone with the updated libphy.a. However the connection is dropped after about 30 seconds. When I do not use tickless, the phone is able to stay connected indefinitely. Any thoughts on what to do here?

EricB-ADI, 2024-11-12:
I am not too sure; I will need to do some tests. What device are you connecting to it with? IOS and Android behave differently, and IOS is not very lenient. The normal cause is sleeping too long, which may put the radio in an in between state of powering on. I am aware of one possible, solution in the PHY, but I will need to test, and if that doesn't work, I need to reach out to our RF team to help with characterization.

sullivanmj, 2024-11-15:
@EricB-ADI Thanks for following up. I'm connecting to a Pixel 3 XL via the LightBlue app.

closed 2025-01-06 by EricB-ADI