analogdevicesinc/msdk #1273: MAX32690 (and others) Reference .ld files

rhempel, 2024-11-17:
Is there a single place where we can find reference linker scripts for MAX32xxx devices?

I am updating my multi-target build system (https://github.com/rhempel/adaptabuild-example) to support the MAX32690FTHR and to simplify the setup, I am using source files from the `hal_adi` repo. 

Here I can find the source for all of the peripheral drivers, C runtime startup, etc.

But there is no default linker script (*.ld) for these devices for either RAM or FLASH based debugging.

I have found a few random.ld files on the web and in the analogdevicesinc group or repos but would be happier if there was a canonical source from AD.

The little PICO debugger works great under Linux and I can communicate with the 32690FTHR board - the driver library builds clean,  I just need a few missing parts like a good linker script to get it all working.

rhempel, 2024-11-18:
I can see that there are .pack files available here:

https://www.keil.arm.com/packs/?q=&pack-search=&vendor=maxim&sort_by=

... but would it be possible to put the contents of each of these .pack files into a separate repo here on GitHub?

ST has separate diver and CMSIS repos for each of their STM32 families, but I still had to reach into the CUBEMX repos to get sample linker scripts - so that's not perfect either.

A good compromise (for me) would be to either have separate repos for the .pack files, or to add the contents to the "hal_adi" repo as a reference.

BobbyCounts, 2024-11-19:
The MAX32690 linker scripts are located here: https://github.com/analogdevicesinc/msdk/tree/main/Libraries/CMSIS/Device/Maxim/MAX32690/Source/GCC

Each device has different linker scripts due to differences in memory size and type. Generally you can find them by amending the above path with the desired part number.

rhempel, 2024-11-21:
Found them, thanks! These are generally device and even project specific so I would generally keep them in a projet specific area anyways.

closed 2024-11-21 by rhempel