analogdevicesinc/msdk #1277: MSDK OpenOCD vs Latest OpenOCD - which coomit is the current AD version From

rhempel, 2024-11-21:
I have verified that he MSDK bulds, flashes, and debugs using VSCode on my Linux machine. So far so good - the vendor supplied toolchain works and was a pleasantly easy install and used VSCode.

@Jake-Carter please extend my thanks to the team that works on MSDK. The tooling, install, and especially the documentation is actually quite good. Isn't it nice when tools like Sphinx are available and are really suited to the task :-)

As nice as MSDK is, I have a toolchain and build process that uses Docker containers for both development (on Linux/Windows/Mac) and CI/CD via GitHub or GitLab automation.

I am trying to sort out why the latest OpenOCD xpack fails to program the 32690 device FLASH - its version is:

`xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-19:50)`

The ADI OpenOCD works fine, but I can't tell exactly which commit it is built from because the version string is:

`Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)`

There's no commit number and no tag so I can't see which version of the original OpenOCD I should be comparing against for differences.

Has ADI made a merge request for changes to support the MAX32xxx devices better been made more recently than 2021? Here is the search for commits with 32xxx in the official OpenOCD repo:

https://repo.or.cz/openocd.git?a=search&h=HEAD&st=commit&s=32xxx&.cgifields=sr

ttmut, 2024-12-16:
Hi @rhempel,

Have you checked ADI's fork of openocd https://github.com/analogdevicesinc/openocd for MAX32xxx related commits?

rhempel, 2024-12-16:
@ttmut yes I have checked that repo for max32xxx commits. It looks like there was some work done to separate the adi-5 and adi-6 debug interfaces, but the latest OpenOCD already has support for those interfaces within the adi-5 files.

I'll have another look to see if the AD/MAX support can be brought into the current openOCD as a smaller set of changes.

Is the current AD OpenOCD tool built from main, or some other branch. If I'm going to do this I want to be sure I'm starting from the right branch :-)

ttmut, 2024-12-17:
I am not much involved in OpenOCD development and I might be completely misunderstanding your query. However, looking at https://github.com/analogdevicesinc/openocd/commits/0.12.0-1.0.0/ suggests that it is https://github.com/openocd-org/openocd/commits/v0.12.0/ plus changes introduced in https://github.com/analogdevicesinc/openocd/commit/3a0e71b8ceea4e040346dcf67366cef5774ab7ce

sihyung-maxim, 2024-12-18:
Hi!

Although I'm not directly involved with the OpenOCD development either, all OpenOCD development for ADI is done within an internal ADI repo (not on GitHub) from which a release is created and subsequently pushed to [analogdevicesinc/openocd](https://github.com/analogdevicesinc/openocd). The OpenOCD binaries that come with the MSDK installer are pulled from the internal repo and, as Tahsin mentioned, match this release here: https://github.com/analogdevicesinc/openocd/releases/tag/0.12.0-1.0.0

I don't know when ADI plans to merge its changes with the official OpenOCD repo, but the links, that Tahsin linked above, are a good place to start!

Let me know if you have any other questions.

rhempel, 2025-01-06:
Thanks for this update - I have been on an extended holiday break and I'm just getting around to checking in on the world :-)

This is a good starting point - at least I can take a delta from the 0.12.0-1.0.0 tag and see what it would take to create a minimal PR for the official OpenOCD repo.

I will update here if/when I have something reasonable to review.

rhempel, 2025-01-13:
@sihyung-maxim @ttmut 

On the official AnalogDevices openocd repo, there are a few different branches/tags that could be a good start. What makes things complicated is code style and spacing changes that just add noise to the diffs.

If we wanted to get the current  0.12.0-1.0.0 tag integrated to the current OpenOCD repo, then I would propose the following:

- Check out https://github.com/analogdevicesinc/openocd/releases/tag/0.12.0-1.0.0
- Create a new branch that cleans up the diffs from the official OpenOCD 0.12.0 branch so that only actual AD changes are visible - no formatting changes
- Build and verify that it works

This is relatively straightforward as the changes are mostly new files and minor modifications to existing OpenOCD files.

Then we need to see if those changes work on the latest OpenOCD master branch:

- Rebase the new, clean diff branch onto the latest OpenOCD commit in master https://github.com/openocd-org/openocd/tree/master
- Build and verify if it works

This is more involved as we will need to rebase a set of changes onto about two years and about 1,000 commits - is it possible, sure. Might it be really, really tedious - yes.

Then we need to figure out if the latest AnalogDevices repo can be brought in as a merge request to OpenOCD.

I understand how things can get out of sync, but as an embedded systems developer, I would like to reduce dependencies on multiple OCD tools and OpenOCD seems to work best for the wide range of Cortex devices I work with.

So far, AnalogDevices/Maxim is the only vendor that is working on their own fork, and I would really like to help out to bring your changes to OpenOCD - at the same time, I'm cautious about spending time on a dead-end effort, or worse, duplicating something that is actually being worked on at AD :-)

rhempel, 2025-01-14:
Change of plans ...

- The current AD release is 0.12.0-1.1.1 at 2f82532f7e94b0f953f2845f49cc405af5abc850
- The first common ancestor with the official OpenOCD is at 01a797af143398c6f3aa1eb6f8949deff4ccf044 - target/mips32: add fpu access support

I will see if the changes between these two points can be separated into C/H files and the tcl config files.

You probably want to keep the existing history for AnalogDevices changes, but if you decide to make a pull request to the OpenOCD upstream, it's better to squash those commits into smaller logical groups so that moving forward the rebase is less painful.

rhempel, 2025-02-05:
@ttmut @sihyung-maxim I have made some progress on this topic. I now have an OpenOCD from the upstream master branch with a minimal set of changes from the AD OpenOCD repo. The branch builds under MSYS2 UCRT on Windows and creates an executable that I can use to debug the MAX32690FTHR board.

There are probably a number of features that I have not ported over, mainly because I now have a "modern" version of OpenOCD that works for me for the MAX32xxx and ALL of the other Cortex M devices that I tend to use.

The current state (with a set of notes for building under MSYS2) is here:

https://github.com/rhempel/openocd/tree/adi-merge

Note, this is not by any means a pull request that OpenOCD is likely to take as-is. The commits should be broken down even further to make it clear what is changing, but I thought this would be better done by AD and of course I could assist if needed.

Feedback welcome.

closed 2025-02-15 by rhempel