analogdevicesinc/msdk #1287: MAX32666 custom board can no longer be flashed with DAPLINK after experimenting with low-power modes

JordanAceto, 2024-12-06:
My team is working on a board with a MAX32666 as the main project. We have a prototype that has been working fine- we've flashed the boards many many times successfully. We are using VSCode for development.

After experimenting with some low-power modes, I believe I have placed my board in the low-power lockout state. I have read this section of the UG, but only to late! https://analogdevicesinc.github.io/msdk/USERGUIDE/#debug-limitations

Attempting to perform the `flash and run` task results in this message in the terminal:
```
The target architecture is set to "armv7e-m".
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
Remote communication error.  Target disconnected.: No error.
```

Attempting to perform the `erase flash` task results in this:
```
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0 (2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to <processor.tools.support@analog.com>
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.0
Info : CMSIS-DAP: Serial# = 04161701d6a74e5a00000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Warn : Connecting DP: stalled AP operation, issuing ABORT
```

Things attempted:
I tried to strategically hit the reset button on the target while connecting several times in hopes that the debugger would connect at just the right time after a reset, with no luck.

I updated the  firmware for my DAPLINK interface with MAX32624PICO image here: https://github.com/analogdevicesinc/max32625pico-firmware-images

I attempted the steps outlined in the UG here: https://analogdevicesinc.github.io/msdk/USERGUIDE/#how-to-unlock-a-microcontroller-that-can-no-longer-be-programmed

I verified that `Automation allowed` is set to `1`.

After copying the `erase.act` file to the DAPLINK, I do not see a `FAIL.TXT` file present in the DAPLINK drive. However, after power cycling the target micro controller I still can't connect to it. The terminal messages are the same.

Is there anything else to try here? Thanks!

ttmut, 2024-12-17:
Can you try doing a release build of the Hello_World example and copy the produced Hello_World.bin file into the DAPLINK drive? To do a release build, right click on the project and click _Build Targets -> Create..._ and create a build target named _release_. Then, right click on the project again and click _Build Targets -> Build..._

JordanAceto, 2024-12-19:
I ended up saving the custom board by power-cycling and flashing at just the right time. It took a bunch of tries but it eventually worked.

Thanks for your suggestion, if this happens again I'll try it out.

Closing this issue, thanks

closed 2024-12-19 by JordanAceto