analogdevicesinc/msdk #1293: MAX32670 I2C Driver causes a Double-Read

Brandon-Hurst, 2024-12-12:
Reading from a peripheral, it appears that the I2C driver for MAX32670 causes a double read. 

## What I Did

Connect an ADXL345 to I2C1 on MAX32670 EVKit. I am running the I2C example, and simply added the below source code before the main test runs...

```C
    // FIXME: For testing
    uint8_t my_adxl_tx[1] = {0x0}; // ADXL ID Register
    uint8_t my_adxl_rx[2] = {0x0};
    mxc_i2c_req_t reqMaster = {
        .i2c = I2C_MASTER,
        .addr = 0x1D, // ADXL345 I2C Addr
        .tx_buf = my_adxl_tx,
        .rx_buf = my_adxl_rx,
        .tx_len = 1,
        .rx_len = 2,
        .callback = NULL,
    };
    I2C_FLAG=1;
    MXC_I2C_MasterTransaction(&reqMaster);
``` 

## Outcome

[image]

The above shows a repeated start and a double read of the register despite only attempting to read once. 

I am trying to use MAX32670 with a No-OS Project that I built and have actually seen this condition cause infinite loops where the I2C bus sends out clocks forever and waits for data that will not come.

## More info 

I dug into the i2c_reva.c driver and it looks like somehow the I2C registers are getting corrupted between when the data comes into the FIFO and when it is actually read. 

Here is when the data has arrived in the FIFO:
[image]

Here's what happens if I single-step a couple assembly instruction past that point:
[image]
[image]

There are no interrupts / DMA enabled, and the data in CPU registers r2/r6 are just some values from RAM. 

This causes the `i2c_reva.c` MasterTransaction function to set the Restart bit and attempt to read again, when no more data will come back from the peripheral.

Brandon-Hurst, 2024-12-14:
May be related to https://github.com/analogdevicesinc/msdk/pull/1211. It looks like the behavior I'm seeing is similar.

dnguye14-adi, 2025-02-17:
Hi, could you please confirm about the i2c configurations?? With rx_len=1, I could not reproduce your signals, but, It is the same if I change the rx_len to 2.

Brandon-Hurst, 2025-03-05:
Hi @dnguye14-adi, thanks, I will re-check this in the next couple of days. Sorry I haven't been able to take another look at it until now.

Brandon-Hurst, 2025-03-05:
Hello,

I went and confirmed this fixes the MasterTransaction function. The configurations from before were correct, e.g. I tested the old vs. new files and saw the same signals as before. The old driver always generates an erroneous double read (the rxLen parameter merely controls the # of bytes read in each of these reads). 

#### Old i2c_reva.c; RxLen=1

[image]

#### Old i2c_reva.c; RxLen=2

[image]

#### New i2c_reva.c; RxLen=1

[image]

I have one change to make to the new proposed i2c_reva.c; the first while loop is unnecessary and adds additional time debugging when this function probably shouldn't be called when the I2C bus is busy in the first place. 

In situations where two parallel tasks might try to access the bus at the same time, those accesses should be controlled with a mutex anyway. I would change this while loop to a simple if statement / return E_BUSY if the busy flag is set. This is also more consistent with how the other PeriphDrivers do it (SPI, UART, etc) and prevents locking up the device under error conditions.

```C
if (i2c->status & MXC_F_I2C_REVA_STATUS_BUSY) {
    return E_BUSY;
}
```

I made this change, recompiled/flashed, and tested this:

[image]

For what it's worth, I also went and made sure the MasterTransactionAsync function work okay. Below is RxLen=1, MasterTransactionAsync:

[image]

dnguye14-adi, 2025-04-10:
Hi,
I'm still wondering about replacing the while loop with checking the busy flag as this is in driver source. If use busy flag checking, then, developer still has to place a while loop in application source. In my opinion, we still keep the while loop in driver section.

Below is my testing with using the while loop: write 1 byte > read 1 byte > write 1 byte > read 1 byte > write 1 byte > read 1 byte > write 1 byte > read 1 byte:

[image]

Brandon-Hurst, 2025-04-10:
Hi @dnguye14-adi, the point is not necessarily to not have a while loop. The point is that the possibility of an _infinite_ while loop can crash the system. If you want to use a while loop in the critical path of a HAL function, you should add a timeout or some other exit condition. 

It is a lot more painful for the user to have a broken system and spend hours debugging our code than it is for them to check the error flags we provide and/or provide their own while loop.

dnguye14-adi, 2025-04-12:
Hi @Brandon-Hurst ,
I updated the PR with your solution, check and return BUSY if the Bus is not ready in the Transaction function.
Passed from my side.

yoshihiko-niino, 2025-05-09:
@dnguye14-adi -san
Please tell me the details.
Please tell me the cause of the double read and the detailed countermeasures.
The same issue is occurring here, so I would like to know the root cause.
Thank you.

dnguye14-adi, 2025-05-09:
@yoshihiko-niino ,
I think the issue is because of executing write and read immediately. That's why I added a check between write and read section:

[image]
 
You should use my latest commit (https://github.com/dnguye14-adi/msdk/commit/57748443d3d1dec396317b9a2fc849b731d33f76), I did a mistake with the merged PR due to a rebasing, It will be updated soon 

Here are the signals after fix:

[image]

Rgds,

yoshihiko-niino, 2025-05-09:
@dnguye14-adi -san
Thank you for your answer.
I'm in charge of hardware, so please explain in more detail.
Why does a read access occur twice like this time when a write and a read are performed immediately? Please explain the mechanism.
Thank you.