analogdevicesinc/msdk #1409: MAX30102 Sensor Stalls After First Reading on MAX78000FTHR

aansrajpoot, 2025-07-10:
Hi Team,

I hope you're doing well.
I’ve successfully integrated the MPU6050 sensor with the MAX78000FTHR board and verified continuous data acquisition.
Now, I'm working on interfacing the MAX30102 heart rate sensor with the same board. The sensor initializes correctly, and I am able to read the first set of IR and RED values successfully. However, after the initial reading, the code stalls, and no further data is retrieved.
I’ve verified the I2C setup and FIFO read logic, and I suspect the issue might be related to interrupt handling or I2C buffering, but I haven’t been able to resolve it.
Could you please help me debug this issue or guide me toward any working examples or suggestions?
Any assistance would be greatly appreciated.
Thank you!

-Below is the attached code file.
[MAX30201.txt](https://github.com/user-attachments/files/21172085/MAX30201.txt)

With the above code, I am getting this output: It gets stuck at uint8_t int_status when the second iteration comes.  
System Init
Checking MAX30102...
MAX30102 detected! PART ID = 0x15
Configuring MAX30102...

 start
 start1
 start2
FIFO read complete with ret = 0
IR: 34, RED: 34
Done

 start

sihyung-maxim, 2025-07-14:
Hi - it's interesting that the code halts at line 148 at `uint8_t int_status`. Could you confirm which fault exception is raised when you encounter the stall?

aansrajpoot, 2025-07-15:
> Hi,
> Thanks for your reply. I hope you're doing well.
>
> I'm currently working on interfacing the MAX30102 heart rate sensor with
> the MAX78000FTHR board. The sensor is successfully detected via I2C, and
> the initialization sequence seems to complete correctly. However, I'm
> encountering a runtime issue during the periodic sampling phase, which
> causes the firmware to stall unexpectedly.
>
> Below is the exact output observed on the serial terminal:
> System Init
> Checking MAX30102...
> MAX30102 detected! PART ID = 0x15
> Initializing MAX30102...
> MAX30102 Initialized.
> Sampling every 1s...
> Starting sample...
> Reading INT_STATUS...
> Reading INT_STATUS (0x00)...
> Reading INT_STATUS2 (0x01)...
> INT_STATUS = 0x40, INT_STATUS2 = 0x00
> Reading FIFO...
> Sample - IR: 102538, RED: 153163
> Sample done. Waiting 1s...
> Starting sample...
> [... 41 lines cut ...]
> Please view the attached file.
>
> Thank you for your time and assistance.
> Regards:
> Aans

Hi,
Here you can view the logs:
got these logs:
Open On-Chip Debugger (Analog Devices 0.12.0-1.0.0-7)  OpenOCD 0.12.0
(2023-09-27-07:53)
Licensed under GNU GPL v2
Report bugs to ***@***.***>
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 0256
Info : CMSIS-DAP: Serial# = 044417011fe28a6500000000000000000000000097969906
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : [max32xxx.cpu] Cortex-M4 r0p1 processor detected
Info : [max32xxx.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for max32xxx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
[max32xxx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x210f0000 pc: 0x10001a9e msp: 0x2001ff38
Warn : Prefer GDB command "target extended-remote :3333" instead of "target
remote :3333"
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x81000000 pc: 0x00002124 msp: 0x20003ff0
Warn : No RTOS could be auto-detected!
Warn : No RTOS could be auto-detected!
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x81000000 pc: 0x00002124 msp: 0x20003ff0
Info : SWD DPIDR 0x2ba01477
[max32xxx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x81000000 pc: 0x00002124 msp: 0x20003ff0

#include "mxc.h"
#include "board.h"
#include "gpio.h"
#include "i2c.h"
#include "stdio.h"

#define I2C_MASTER       MXC_I2C1
#define MAX30102_ADDR    0x57
#define REG_PART_ID      0xFF
#define REG_FIFO_DATA    0x07
#define REG_FIFO_WR_PTR  0x04
#define REG_FIFO_RD_PTR  0x06
#define REG_OVF_COUNTER  0x05
#define REG_INT_STATUS   0x00

#define I2C_IS_BUSY(i2c)   ((i2c->status & MXC_F_I2C_STATUS_BUSY) != 0)

void delay_ms(int ms) {
    MXC_Delay(MXC_DELAY_MSEC(ms));
}

void clear_i2c_flags(void) {
    unsigned int flags0 = 0, flags1 = 0;
    MXC_I2C_GetFlags(I2C_MASTER, &flags0, &flags1);
    MXC_I2C_ClearFlags(I2C_MASTER, flags0, flags1);
}

int i2c_transaction_timeout(mxc_i2c_req_t* req, uint32_t timeout_ms) {
    uint32_t start = MXC_TMR0->cnt;
[... 197 lines cut ...]
        return -1;
    }
    MXC_I2C_SetFrequency(I2C_MASTER, 400000);

    printf("Checking MAX30102...\n");
    if (max30102_read_reg(REG_PART_ID, &part_id) != E_NO_ERROR) {
        printf("ERROR: Failed to read PART ID\n");
        return -1;
    }
    printf("MAX30102 detected! PART ID = 0x%02X\n", part_id);

    if (max30102_init() != 0) {
        printf("ERROR: Sensor initialization failed\n");
        return -1;
    }

    printf("Sampling every 1s...\n");

    while (1) {
        printf("Starting sample...\n");
        read_max30102_sample();
        printf("Sample done. Waiting 1s...\n");
        delay_ms(1000);
    }
}

aansrajpoot, 2025-07-28:
Hi,
I hope you're doing well.
Although I don't currently have access to a logic analyzer, I’ve performed a register dump of the MAX30102 sensor’s I2C registers before and after the failure point using the MAX78000FTHR board.

I’ve attached the code I used along with the console output below. The key point observed is that the INT_STATUS register initially shows 0x40 (indicating new FIFO data available), but immediately after that it changes to 0x00, even though no data has yet been read—suggesting the interrupt flag may be getting cleared unexpectedly or prematurely.

In addition, the overflow counter (REG 0x05) consistently shows 0x01, which triggers the “FIFO Overflow detected” logic—resulting in a FIFO pointer reset. This behavior appears to repeat each second, meaning samples are being acquired, but interrupt handling or data clearing is not stable.
Please have a look and let me know your insights.

[output.txt](https://github.com/user-attachments/files/21474688/output.txt)

[code.txt](https://github.com/user-attachments/files/21474719/code.txt)