analogdevicesinc/msdk #1420: I2C double read err (RESTART-mode)

yoshihiko-niino, 2025-08-04:
Hello.

I'm reposting this as a new issue.
#1394 mentions an I2C double read error.
This issue was resolved by replacing i2c_reva.c, but that alone
doesn't seem to be a complete solution. I'll explain it below.

The "START" → "STOP" condition alone is not a problem, but when communicating using "RESTART," the address is incremented and data is output at the subsequent stage.

(Example)
START → Address 0x56 (W)
→ Register Address: 0x10,0x00
RESTART → Address 0x56 (R)
→ 0x07,0x6A ← As expected
RESTART → Address 0x56 (R)
→ 0x00,0x01 ← For some reason, the data at the next address is displayed.
STOP

I think some kind of workaround is needed, such as preventing the BUSY check from passing when "RESTART" occurs.

I tried the following solution, which distinguishes between START and RESTART and ignores RESTART. This appears to resolve the issue. However, I'm not sure if there are any side effects.

Fixed lines 883 and onward.
---------------------------------------------------------------------------------
/* ************************************************************************* */
/* Transaction level functions                                               */
/* ************************************************************************* */

int MXC_I2C_RevA_MasterTransaction(mxc_i2c_reva_req_t *req)
{
    mxc_i2c_reva_regs_t *i2c = req->i2c; // Save off pointer for faster access
    unsigned int written = 0;
    unsigned int read = 0;

    int is_first_start = !(req->restart);  // restart==0 のとき最初のSTARTとみなす

    if (req->addr > MXC_I2C_REVA_MAX_ADDR_WIDTH) {
        return E_NOT_SUPPORTED;
    }

    if (MXC_I2C_GET_IDX((mxc_i2c_regs_t *)i2c) < 0) {
        return E_BAD_PARAM;
    }

    if (!(i2c->ctrl & MXC_F_I2C_REVA_CTRL_MST_MODE)) {
        return E_BAD_STATE;
    }

    // Check and return BUSY only if this is the first START
    if (is_first_start && (i2c->status & MXC_F_I2C_REVA_STATUS_BUSY)) {
        return E_BUSY;
    }

    // if(!read | write)
    //  Start
    //  send addr w/ write bit
    // if(Write)
    //  send tx_len data
    //  return if error (or NACK)
    // if(Read)
    //  if(Write)
    //   send restart
    //  else
    //   send start
    //  send addr w/ read bit
    //  read rx_len bytes acking all
    // stop or restart
    // return good or error

    MXC_I2C_ClearFlags((mxc_i2c_regs_t *)i2c, MXC_I2C_REVA_INTFL0_MASK,
                       MXC_I2C_REVA_INTFL1_MASK); // Clear all I2C Interrupts
    MXC_I2C_ClearTXFIFO((mxc_i2c_regs_t *)i2c);
    MXC_I2C_ClearRXFIFO((mxc_i2c_regs_t *)i2c);
    i2c->inten0 = 0;
    i2c->inten1 = 0;

    if ((req->rx_len == 0) || (req->tx_len != 0)) {
        // Load the slave address with write bit set
        i2c->fifo = (req->addr << 1) & ~0x1;
        i2c->mstctrl |= MXC_F_I2C_REVA_MSTCTRL_START;
    }

    while (req->tx_len > written) {
        if (i2c->intfl0 & MXC_F_I2C_REVA_INTFL0_TX_THD) {
            written += MXC_I2C_WriteTXFIFO((mxc_i2c_regs_t *)i2c, &req->tx_buf[written],
                                           req->tx_len - written);
            i2c->intfl0 = MXC_F_I2C_REVA_INTFL0_TX_THD;
        }

        if (i2c->intfl0 & MXC_I2C_REVA_ERROR) {
            req->tx_len = written;
            MXC_I2C_Stop((mxc_i2c_regs_t *)i2c);
            return E_COMM_ERR;
        }
    }

    MXC_I2C_ClearFlags((mxc_i2c_regs_t *)i2c,
                       MXC_F_I2C_REVA_INTFL0_DONE | MXC_F_I2C_REVA_INTFL0_RX_THD, 0);

    if (req->rx_len != 0) {
        if (req->rx_len > MXC_I2C_REVA_MAX_FIFO_TRANSACTION) {
            i2c->rxctrl1 = 0;
        } else {
            i2c->rxctrl1 = req->rx_len; // 0 for 256, otherwise number of bytes to read
        }

        MXC_I2C_Start((mxc_i2c_regs_t *)i2c); // Start or Restart as needed

        while (i2c->mstctrl & MXC_F_I2C_REVA_MSTCTRL_RESTART) {}

        i2c->fifo = (req->addr << 1) | 0x1; // Load slave address with read bit.
    }

    if ((req->rx_len != 0) && (req->tx_len != 0)) {
        while (!(i2c->intfl0 & MXC_F_I2C_REVA_INTFL0_DONE)) {}
        // Wait for Transaction to finish

        i2c->intfl0 |= MXC_F_I2C_REVA_INTFL0_DONE;
    }

    while (req->rx_len > read) {
        if (i2c->intfl0 & (MXC_F_I2C_REVA_INTFL0_RX_THD | MXC_F_I2C_REVA_INTFL0_DONE)) {
            read +=
                MXC_I2C_ReadRXFIFO((mxc_i2c_regs_t *)i2c, &req->rx_buf[read], req->rx_len - read);
            i2c->intfl0 = MXC_F_I2C_REVA_INTFL0_RX_THD;
        }

        if (i2c->intfl0 & MXC_I2C_REVA_ERROR) {
            req->rx_len = read;
            MXC_I2C_Stop((mxc_i2c_regs_t *)i2c);
            return E_COMM_ERR;
        }

        if ((i2c->intfl0 & MXC_F_I2C_REVA_INTFL0_DONE) && (req->rx_len > read) &&
            (MXC_I2C_RevA_GetRXFIFOAvailable(i2c) == 0)) {
            if ((req->rx_len - read) > MXC_I2C_REVA_MAX_FIFO_TRANSACTION) {
                i2c->rxctrl1 = 0;
            } else {
                i2c->rxctrl1 = (req->rx_len - read); // 0 for 256, otherwise number of bytes to read
            }

            i2c->mstctrl |= MXC_F_I2C_REVA_MSTCTRL_RESTART;
            i2c->intfl0 = MXC_F_I2C_REVA_INTFL0_DONE;
            i2c->fifo = (req->addr << 1) | 0x1; // Load slave address with read bit.
        }
    }

    if (req->restart) {
        i2c->mstctrl |= MXC_F_I2C_REVA_MSTCTRL_RESTART;
    } else {
        i2c->mstctrl |= MXC_F_I2C_REVA_MSTCTRL_STOP;

        while ((i2c->mstctrl & MXC_F_I2C_REVA_MSTCTRL_STOP)) {}
        // Wait for Transaction to finish
    }

    while (!(i2c->intfl0 & MXC_F_I2C_REVA_INTFL0_DONE)) {}
    // Wait for Transaction to finish

    i2c->intfl0 = MXC_F_I2C_REVA_INTFL0_DONE | MXC_F_I2C_REVA_INTFL0_STOP;

    if (i2c->intfl0 & MXC_I2C_REVA_ERROR) {
        return E_COMM_ERR;
    }

    return E_NO_ERROR;
}
---------------------------------------------------------------------------------

Thank you.

dnguye14-adi, 2025-08-11:
Hello,
Can you also attach your example code in the main, and then I can reproduce the issue from my side.

Rgds,

ttmut, 2025-08-19:
Hi @yoshihiko-niino -san,

Isn't this how most I2C devices operate? After each read operation, the internal address pointer is incremented, so for the next read operation, the value from the incremented address is returned.

dnguye14-adi, 2025-09-04:
hi @yoshihiko-niino -san, 
i saw that your code can only work with the MAX78000, but was failed on another boards (eg. MAX32690 ). So, i need more time to get a solution for all the boards.

brgs,

yoshihiko-niino, 2025-09-04:
Hi,@dnguye14-adi-san

Thank you for your reply.
I understand the content.
I apologize for the inconvenience, but please continue to consider this in order to find the optimal solution.
Thank you.

Best Regards,
Yoshihiko Niino